roleoroleo/yi-hack-Allwinner-v2 #1: Offering 2 cams for testing

pep4n, 2020-10-08:
Hey @roleoroleo 
I have 2 of these cams here which I could offer for testing. 
Also would love to support. Only my dev skills are near zero. 
Id also be available for a vid call if that could be of help.
Let me know, 
p

roleoroleo, 2020-10-08:
I could sure use.

pep4n, 2020-10-08:
Ah sry I rather meant remotely. 
I guess sending these across Europe is more expensive then what I paid on amazon ,)

roleoroleo, 2020-10-08:
Remotely it's ok.

luisglezi, 2020-10-08:
I have also 2 brand new cameras that can be used for testing

pep4n, 2020-10-08:
> 
> 
> Remotely it's ok.

ok cool. let me know how we should do this!

pep4n, 2020-10-09:
I'd be available tonight for some remote testing. 
Cheers,
P

roleoroleo, 2020-10-09:
What UTC offset?
:)

luisglezi, 2020-10-09:
I will not be available until Thu next week. I am GMT+2 (Spain)

pep4n, 2020-10-09:
> 
> 
> What UTC offset?
> :)

I'm in CEST. 
So atm GMT + 2. 

Would be available from around 8pm. 
What App would you like to use?

roleoroleo, 2020-10-09:
I'm gmt+2 too.
But tonight I'm not at home.
Sorry.

roleoroleo, 2020-10-09:
About the application...
Are you able to publish ssh with a temporary port forwarding?
Maybe on a non-standard port?

pep4n, 2020-10-09:
> 
> 
> I'm gmt+2 too.
> But tonight I'm not at home.
> Sorry.

no worries & happy Friday! 
Only we need to fix something so I can block it. 
I have a wife and two kids who book slots very fast ,)  

for SSH ports I need to check. 
I am in a double NAT network and only have direct access to one of the routers. 
Would teamviewer or similar also be an option?

roleoroleo, 2020-10-09:
I can understand.
:)
I have two wifes and three kids!

Teamviewer is ok for me.

pep4n, 2020-10-09:
ahaha that would sure be too much for me ,) 

Lets see if we find a time.

pep4n, 2020-10-18:
I have some time today...

PS: What applications would you need? 
To not steal more of your time I would already install them before we teamview...

roleoroleo, 2020-10-18:
Sorry, I read the message too late.
Are you online?

pep4n, 2020-10-18:
No worries, need to cook dinner now. 
Would be available from 8 on...

roleoroleo, 2020-10-18:
We will never make it... :)

HuangRicky, 2020-10-21:
Hi @roleoroleo  I have a [Yi Outdoor 1080p](https://www.amazon.com/YI-Waterproof-Surveillance-Detection-Deterrent/dp/B01CW49AGG/ref=sr_1_3?dchild=1&keywords=yi+outdoor+camera&qid=1603244323&sr=8-3)  

Does this fw work on this cam? I can also provide it for testing if you tell me what I need to do: I assume I shall flash it and then expose the ssh thru port-forwarding? Thanks!

HuangRicky, 2020-10-21:
the serial number starts with 9CUS

closed 2020-10-24 by pep4n