roleoroleo/yi-hack-Allwinner-v2 #2: Solder wires for uart bridge

lassieee, 2020-10-08:
This is not so much an actual issue, but more to get confirmation about where to solder the wires to attach a uart bridge. 
Can you please tell if I'm correct in the location for the wires? 
[image]

danielk117, 2020-10-08:
Yes you are right. 
In theory, you don't even have to solder. You just need someone with a steady hand or some skill with tape. 😉

lassieee, 2020-10-08:
attached is my uart logging (obtained with tremendous tape skills ;-)), perhaps it's useful.

[screen_output.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5350886/screen_output.txt)

roleoroleo, 2020-10-08:
Are you able to login with a root shell?

lassieee, 2020-10-08:
I couldn't really test.. I had to keep my fingers on the tape lol. If I feel lucky tomorrow, I might give soldering a try and let you know :-)

lassieee, 2020-10-09:
unfortunately no shell, or I might be doing something wrong. I tried with 'screen /dev/ttyUSB0 115200' and with putty.
I wasn't able to solder the wires. The little copper dots on the board are just too damn tiny! I lack proper tools and skills. The solder doesn't stick to the copper dots and I don't have any flux, nor have I ever used it.

roleoroleo, 2020-10-09:
If you have no shell, we have to stop the boot process and work with u-boot.
Like the previous model.

roleoroleo, 2020-10-24:
@lassieee 
Could you check it the wiki I wrote is ok?
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Unbrick-the-cam

Could I add your image posted above?

lassieee, 2020-10-25:
Yes, sure you can use the picture. Happy to be of any help :-)

At step "2: List the content of the SD card:", perhaps you could change the output to include the files that are actually on the sd card for this version:
```
=> fatls mmc 0
            record/
     2502   lower_half_init.sh
            newbackup/
            yi-hack/
  1245184   mtdblock4.bin
            Factory.done/
            log/

2 file(s), 5 dir(s)
```

The flash commands are what I've used and work great. I see you already added the v2 partition table.

At step "8: Reset and cross your fingers." I would change it to:
8: Reset and cross your fingers:
`=> reset`

p.s. with the uart wires properly attached, I do have a root shell on the serial console after the cam boots.

roleoroleo, 2020-10-25:
Ok, thanks.

closed 2020-12-27 by roleoroleo