roleoroleo/yi-hack-Allwinner-v2 #5: 0.1.0 boot loop

pep4n, 2020-10-24:
Hey there, 

camera is send to bootloop when trying to boot up 0.1.0: 
Boots up with yellow LED until it is stable blue. 
Does the click sound once. 
HTTP is reachable for a few seconds. 
Then cam reboots. 

Just applied 0.0.8 which is booting up and seems not to reboot (few minutes now). 

Cheers,
P

damanti-me, 2020-10-24:
I can confirm this.

Blutnovsky, 2020-10-24:
I have the same issue

roleoroleo, 2020-10-24:
I remove the release temporarily.

c3kkos, 2020-10-24:
Where the webui stores settings? May they interfere somehow with services booting up?

roleoroleo, 2020-10-24:
The webui stores the configuration in /tmp/sd/yi-hack/etc

I compared 0.0.8 and 0.1.0 and I can't understand the problem.
Remember that I don't own this cam :)
I need help.

c3kkos, 2020-10-24:
Are there any watchdogs active?

roleoroleo, 2020-10-24:
Only if rmm is not running.
But I think it's not possible.

roleoroleo, 2020-10-26:
Could you try this?
[y21ga_0.1.0.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5440642/y21ga_0.1.0.tar.gz)

roleoroleo, 2020-10-26:
Could you try this?
[y21ga_0.1.1.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5440647/y21ga_0.1.1.tar.gz)

pep4n, 2020-10-27:
> 
> 
> Could you try this?
> [y21ga_0.1.1.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5440647/y21ga_0.1.1.tar.gz)

fixed the boot loop! :) 
I have the feeling the cam, or at least HTTP is work more smoothly. 
RTSP delay via VLC: 2sec. 
no artefacts or other video related issues

issues now: 
- snapshots (high: greenish broken / low: not loading) 
- no audio in RTSP stream 

I guess this issue can be closed for now.

closed 2020-10-27 by roleoroleo