roleoroleo/yi-hack-Allwinner-v2 #6: Mqtt not working on 0.0.8

c3kkos, 2020-10-24:
Hi everybody, i'm testing mqtt functionality of yi hack with home assistant, but i' stuck. I cannot get yi hack cam to connect to hassio mqtt broker (mosquitto)

I tried connection with other test clients (mqtt dash android), and i can see succesful logins into the logs.

No logs entries for the yihack cam.

Running 0.0.8 yihack4 allwinner v2 on a yi 1080 camera (BFUSY44x.... Serial)

roleoroleo, 2020-10-25:
Do you see birth messages at boot time?

1st step is check if the daemon starts:

- Check if is running using ps command
- If not, try to run it manually:
```
. /tmp/sd/yi-hack/script/env.sh
mqttv4 &
```

2nd step if mqttv4 is running:

- Subscribe to the topic and check birth message when the cam boots (If you use ha: integrations->mqtt->configure).
- You shoud see online message.

damanti-me, 2020-10-25:
Hey @roleoroleo. I can confirm the birth online message. Thank you for your support!!! 
We probably need to set mqtt for yi cameras manually in the config...

roleoroleo, 2020-10-25:
If you configure mqtt ftom the web page, does mqttv4.conf change?
The path is /tmp/sd/yi-hack/etc

c3kkos, 2020-10-25:
Ok i lack the binary! I'll go to 0.1.0 release... I'll soon check and report here.

damanti-me, 2020-10-25:
> 
> 
> If you configure mqtt ftom the web page, does mqttv4.conf change?
> The path is /tmp/sd/yi-hack/etc

Yes. the mqttv4.conf canges!

c3kkos, 2020-10-25:
[code]
root@yicam1:/# ./tmp/sd/yi-hack/script/env.sh

root@yicam1:/tmp/sd/yi-hack/bin# ls
base64           h264grabber      ipc_read         rRTSPServer
dbclient         h264grabber_h    mosquitto_pub    scp
dropbearconvert  h264grabber_l    mosquitto_sub    ssh
dropbearkey      hostname         mqttv4           tar
dropbearmulti    imggrabber       onvif_srvd       wsdd
framefinder      ipc_cmd          proccgi
gzip             ipc_multiplexer  proxychains4

root@yicam1:/tmp/sd/yi-hack/bin# mqttv4 &
root@yicam1:/tmp/sd/yi-hack/bin# -ash: mqttv4: not found

[/code]

roleoroleo, 2020-10-25:
You have to add a space bwtween . and / in the env command

c3kkos, 2020-10-26:
Sorry for that, newbie post.

Yes, connected. here's the logs:

root@yicam1:~# Starting mqttv4 v0.0.3
Trying to connect... Client yicam1 sending CONNECT
.Client yicam1 received CONNACK (0)
Client yicam1 sending PUBLISH (d0, q1, r1, m1, 'homeassistant/status', ... (6 bytes))
. connected!
Client yicam1 received PUBACK (Mid: 1)
Client yicam1 sending PINGREQ
Client yicam1 received PINGRESP

for HA side, we need to set the right birth message to trigger the auto discovery, this is HA business, not roleo's fw.

Thanks.

Anyway, "the issue" now is that we have to start mqtt manually in 0.0.8

roleoroleo, 2020-10-26:
> Anyway, "the issue" now is that we have to start mqtt manually in 0.0.8

Why?
When you start the cam, is it running?

pep4n, 2020-10-27:
> Hi everybody, i'm testing mqtt functionality of yi hack with home assistant, but i' stuck. I cannot get yi hack cam to connect to hassio mqtt broker (mosquitto)
>
> I tried connection with other test [...]

Do you still experience this in 0.1.1?

damanti-me, 2020-10-28:
I'm on 0.1.2 y21ga. I get no motion mqtt alert (state changing binary_sensor) with the settings provided in the wiki for the home assistant part. 
- Mqtt set up in the settings of the camera
- Successfully connected to broker on home assistant 
- online state during camera boot 

Does anyone using mqtt motion binary sensor and got it work?

mcanteri, 2020-10-29:
Yes, me. Here is my configuration, saved in the separated file /config/packages/yicam_motion.yaml

binary_sensor:
  - platform: mqtt
    name: "Yi Camera Motion Detection"
    state_topic: "yicam/motion_detection"
    payload_on: "motion_start"
    payload_off: "motion_stop"
    qos: 1
    off_delay: 5
  - platform: mqtt
    name: "Yi Camera AI Human Detection"
    state_topic: "yicam/ai_human_detection"
    payload_on: "human_start"
    payload_off: "human_stop"
    qos: 1
    off_delay: 5
  - platform: mqtt
    name: "Yi Camera AI Baby Crying"
    state_topic: "yicam/baby_crying"
    payload_on: "crying"
    qos: 1
    off_delay: 5
  - platform: mqtt
    name: "Yi Camera Motion Files"
    state_topic: "yicam/motion_files"
    value_template: "{% if value_json.files %}ON{% else %}OFF{% endif %}"
    qos: 1
    off_delay: 5

On the camera side, enable and let all parameters to default values. Obviously you have to add user and passwd and host. I think that's it.

mcanteri, 2020-10-29:
BTW, if you try enabling human detection or baby crying, mqtt is not working, but if you like give it a shot.

roleoroleo, 2020-10-29:
The last version also supports mqtt camera.

damanti-me, 2020-10-29:
@mcanteri @roleoroleo thank you for the support. Ill give it another try...

mcanteri, 2020-10-29:
> The last version also supports mqtt camera.

@roleoroleo the configuration of the camera support already now that feature ?

roleoroleo, 2020-10-30:
The engine (mqttv4) supports it.
The web interface is commited but not released:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/ff3154c86ebd4394f9c54fcdc26850280ddd7562

c3kkos, 2020-10-31:
> > Anyway, "the issue" now is that we have to start mqtt manually in 0.0.8
> 
> Why?
> When you start the cam, is it running?

SO sorry for being so late.
NO i can confirm that mqtt4v is up and running after reboot.

I'm still on 0.0.8, i'll upgrade soon

> Yes, me. Here is my configuration, saved in the separated file /config/packages/yicam_motion.yaml
>
> ```
> binary_sensor:
>   - platform: mqtt
>     name: "Yi Camera Motion Detection"
>     state_topic: [...]

i got that config in my HA but not working -_- i've to debug with a mqtt client 

AND... can u write up the camera config for live viewing in lovelace homepage?

c3kkos, 2020-10-31:
ok so i decide to write things here to soon without further testing.

Here's the status:

i got 0.0.8 with mqtt4v running. Yicam connects to the HA's mqtt broker and i can see it.

I'm now investigating upon messages being sent by the little cam. There are none. no birth, no will, no statuses.

I've got by apt mosquitto clients helpers and with them i checked if mqtt side is working and.... i can confirm yes it works.

```
```
$ mosquitto_pub -h 192.168.1.177 -p 1883 -u * -P * -t yicam1 -d -m test
Client mosqpub|11700-garage sending CONNECT
Client mosqpub|11700-garage received CONNACK (0)
Client mosqpub|11700-garage sending PUBLISH (d0, q0, r0, m1, 'yicam1', ... (4 bytes))
Client mosqpub|11700-garage sending DISCONNECT
```

this is a message sent with mosquitto_pub at the broker, with the same prefix set in the webui "yicam1"

and this is the message recieved by mosquitto_sub, opened in a second terminal:

$ mosquitto_sub -h 192.168.1.177 -p 1883 -u * -P * -t yicam1 -d
Client mosqsub|11237-garage sending CONNECT
Client mosqsub|11237-garage received CONNACK (0)
Client mosqsub|11237-garage sending SUBSCRIBE (Mid: 1, Topic: yicam1, QoS: 0)
Client mosqsub|11237-garage received SUBACK
Subscribed (mid: 1): 0
Client mosqsub|11237-garage sending PINGREQ
Client mosqsub|11237-garage received PINGRESP
Client mosqsub|11237-garage received PUBLISH (d0, q0, r0, m0, 'yicam1', ... (4 bytes))
test
```

mqtt is working  but yicam is sending none.

roleoroleo, 2020-10-31:
Try to upgrade to the last version.
I don't remember if I fixed something.

c3kkos, 2020-10-31:
Can I update on the fly? Like use scp to write files onto the sd and reboot?

roleoroleo, 2020-10-31:
I don't remember if online update works in your version.
And I don't remember if backup/restore config works.
Extract the sd and copy the content of the new archive.
And reconfigure the cam.

mcanteri, 2020-10-31:
I'm quite a beginner to mosquitto anyway I found this tool useful:
MQTT-Explorer-0.4.0-beta1.AppImage

c3kkos, 2020-11-01:
Hi there,

just updated the cam to 0.1.2 roleo's release.

I can confirm now that mqtt is working.

mcantieri, i'm using the nice and free **mqtt-fx** in my windows desktop, subscribing to "#" seeing all the messages being pushed through the broker.

This said, i added this piece of config in home assistant:

```
  - platform: mqtt
    name: "Yi Camera Motion Status"
    state_topic: "yicam1/status"
    payload_on: "online"
    payload_off: "offline"
    qos: 1

```

A nice feature to have, gives u an immediate feedback of cam status if u add that to a card, in lovelace.

Basically that binary sensor goes on when the birth message is given by the cam (which by default has the retain flag ON)

Cam goes UP, status "online" is being pushed by mqtt. Cam is online for homeassistant. For the message is retained, the status can be aquired from HA even cam was up before HA is.

For now on, no status updates are coming. mqtt cam side is configured to send keepalives; this means cam and mqtt broker "ping each other" to check if there's link beetween them, but NO mqtt messages are send.

This is bad, binary sensor won't work... There's a way to tell CAM to send status updates within X seconds?

We can then set that update, let's say every 60 seconds, and put in the above configuration the "off_delay: 120"

mcanteri, 2020-11-01:
The status message, in Home Assistant, as you say it isn't much useful, but the motion one it is and (I'm working on it) the motion_files too.
With those binary you can build an easy automation that use camera.record and record a video to a file, as long as you like and then send it to your Telegram chat.

mcanteri, 2020-11-01:
> The engine (mqttv4) supports it.
> The web interface is commited but not released:
> [ff3154c](https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/ff3154c86ebd4394f9c54fcdc26850280ddd7562)

@roleoroleo have to wait the next release also, my local build process is no more working after the last release.

c3kkos, 2020-11-01:
Mcantieri can you write down the camera config you have in your HA? The one in config.yaml and lovelace-ui too

mcanteri, 2020-11-01:
Here the starting page:
https://www.home-assistant.io/integrations/camera.ffmpeg/

In configuration.yaml
`
ffmpeg:
stream:
camera:
  - platform: ffmpeg
    input: rtsp://192.168.1.132/ch0_0.h264
    name: living_room_camera
#    extra_arguments: -rtsp_transport tcp -analyzeduration 15 (to use when net connection is not so good)
`
In the GUI you just have to add a Picture Entity Card and put info about your camera, like:
camera.living_room_camera

c3kkos, 2020-11-01:
Thx. Nothing strange, its like the one i already have but... No luck. No snapshot preview in HA and even the stream is soo slow.. 10 seconds loading before video appears.. and i got no services on cam but mqtt and high res rtsp

mcanteri, 2020-11-02:
Have you enabled the swap space feature ? How's the connection quality presented by the camera page ?
My configuration is with both high and low streams enabled and camera disconnected from cloud. 
Changing which steam to enable or just turning off and on camera in configuration, I saw one time a change in behaviour.

c3kkos, 2020-11-02:
No swap enabled. It harms the SD, so my will is to keep it off. Of course, if swap helps the little cam and improve the performance i'll turn it ON.

I got only high res stream enabled.

Conn quality is 80% stable. The AP is 4 meters away, no issues there.

I'll try swap and let you know. Thank you

timknowlden, 2020-12-21:
> Here the starting page:
> https://www.home-assistant.io/integrations/camera.ffmpeg/
>
> In configuration.yaml
>
> ```
> ffmpeg:
> stream:
> camera:
>   - platform: ffmpeg
>     input: [...]

Hi,
 I have integrated the rtsp live view into HA with ffmpeg, no sound though, have you had any joy with the audio?

mcanteri, 2020-12-21:
> Hi,
> I have integrated the rtsp live view into HA with ffmpeg, no sound though, have you had any joy with the audio?

No, the sound never worked on HA to me and I think is an issue on the HA side. Probably should open an issue there.

closed 2020-12-27 by roleoroleo