roleoroleo/yi-hack-Allwinner-v2 #7: y21ga 0.1.0 issues

mcanteri, 2020-10-25:
Snapshots no more works, both streams and page.
Streams miss audio, video works both with direct player (i.e. mplayer) and integrated with HA.
MQTT motion event works but only if AI Human Detection and/or Baby crying detect are disabled.
Minor issue, passwd cannot be changed because /etc/passwd is on readonly file system.

In kernel log I found these lines that could be interesting:
[    8.547320] snddaudio snddaudio0: codec_dai set sysclk failed
[    8.547331] snddaudio snddaudio0: codec_dai set set_pll failed.
[    8.547337] snddaudio snddaudio0: codec dai set fmt failed
[    8.547355] snddaudio snddaudio0: codec_dai set clkdiv failed
[    8.735326] snddaudio snddaudio0: codec_dai set sysclk failed
[    8.735337] snddaudio snddaudio0: codec_dai set set_pll failed.
[    8.735343] snddaudio snddaudio0: codec dai set fmt failed
[    8.735360] snddaudio snddaudio0: codec_dai set clkdiv failed
[    8.855979] s_fmt set width = 1936, height = 1096
[    8.915069] [sp2305_mipi] error, -------- sensor_s_ctrl : 9963797 --------
[    8.933761] into set sensor hfilp the value:0 
[    8.935381] [VIN]sensor_flip_option:isp0 reset!!!
[    8.935473] [sp2305_mipi] error, -------- sensor_s_ctrl : 9963796 --------
[    8.952290] into set sensor hfilp the value:0 
[    8.953696] [VIN]sensor_flip_option:isp0 reset!!!
[    8.954624] cedar_ve: VE real_fre=300000000

roleoroleo, 2020-10-26:
> Snapshots no more works, both streams and page.

Ok, I will check it.

> Streams miss audio, video works both with direct player (i.e. mplayer) and integrated with HA.

This is strange. No changes between 0.0.8 and 0.1.0 about audio.

> MQTT motion event works but only if AI Human Detection and/or Baby crying detect are disabled.

This is a know issue in other models as well.

What about boot loop?
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/5

damanti-me, 2020-10-26:
> > Streams miss audio, video works both with direct player (i.e. mplayer) and integrated with HA.
> 
> This is strange. No changes between 0.0.8 and 0.1.0 about audio.
_y21ga on 0.0.8_
Audio doesnt work for me at all (RTSP, RTSP audio). Tried to stream on different plattforms like VLC and HA. Just the recorded videos directly on the SD Card does have audio.

pep4n, 2020-10-26:
> Snapshots no more works, both streams and page.
> Streams miss audio, video works both with direct player (i.e. mplayer) and integrated with HA.
> MQTT motion event works but only if AI Human Detection [...]

how come your cam is not sent to bootloop as the others? Sure using 0.1.0?

mcanteri, 2020-10-26:
> > Streams miss audio, video works both with direct player (i.e. mplayer) and integrated with HA.
> 
> This is strange. No changes between 0.0.8 and 0.1.0 about audio.

For me it wasn't working in previous either. Said in other thread too.

mcanteri, 2020-10-26:
> how come your cam is not sent to bootloop as the others? Sure using 0.1.0?

I saw the post about boot loop and I assumed, looking on changes checked in, that the significant changes are in the folder yi-hack, so updated only that folder.

roleoroleo, 2020-10-26:
> I saw the post about boot loop and I assumed, looking on changes checked in, that the significant changes are in the folder yi-hack, so updated only that folder.

Eplain me better.
You extracted yi-hack folder from 0.1.0 version and copied it on top of 0.8.0.
Right?

mcanteri, 2020-10-26:
Yes, deleted the old folder on the SD card and extracted the new one from the archive. Then I restore the backup of configuration and reboot again. All binaries are under yi-hack, only the hack is outside for what I understood.
I'm building the archive from source, previously I was using the published one.

pep4n, 2020-10-26:
> Yes, deleted the old folder on the SD card and extracted the new one from the archive. Then I restore the backup of configuration and reboot again. All binaries are under yi-hack, only the hack is [...]

MacGyver is this you? 
,)

mcanteri, 2020-10-27:
New release out, I close this.

closed 2020-10-27 by mcanteri