roleoroleo/yi-hack-Allwinner-v2 #8: Where to get the latest files?

LukasVFL99, 2020-10-26:
I'm looking for the latest files for the BFUS 9.0xxx?
Any help is appreciated

roleoroleo, 2020-10-26:
The hack is in beta testing.
No releases at the moment.

fklement, 2020-10-26:
@roleoroleo I just wanted to build the firmware using Debian 10 but somehow the compilation fails. 
Couldn't you just release the beta firmware files for testing purposes?

roleoroleo, 2020-10-26:
Beta testing issue...

fklement, 2020-10-26:
Hm? Not sure what you want to tell me?
Currently, the compilation is running under ubuntu...
If everything works out @LukasVFL99 I can upload the firmware for you.

roleoroleo, 2020-10-26:
Sorry, you can find beta releases in this post:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/5#issuecomment-716736627
or in this issue:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/3

roleoroleo, 2020-10-26:
I'm currently use debian 9 to build the firmware.

mcanteri, 2020-10-27:
ok, as never said, with the new release 0.1.2 now go in error.

roleoroleo, 2020-10-27:
What?
The build process?

mcanteri, 2020-10-27:
Yes. It ends like this:
`mv -f .deps/soapcpp2-symbol2.Tpo .deps/soapcpp2-symbol2.Po
gcc -DHAVE_CONFIG_H -I. -I../..    -DWITH_BISON -DWITH_FLEX  -DSOAPCPP2_IMPORT_PATH="\"/usr/local/share/gsoap/import\"" -DLINUX -g -O2 -MT soapcpp2-error2.o -MD -MP -MF .deps/soapcpp2-error2.Tpo -c -o soapcpp2-error2.o `test -f 'error2.c' || echo './'`error2.c
mv -f .deps/soapcpp2-error2.Tpo .deps/soapcpp2-error2.Po
gcc -DHAVE_CONFIG_H -I. -I../..    -DWITH_BISON -DWITH_FLEX  -DSOAPCPP2_IMPORT_PATH="\"/usr/local/share/gsoap/import\"" -DLINUX -g -O2 -MT soapcpp2-init2.o -MD -MP -MF .deps/soapcpp2-init2.Tpo -c -o soapcpp2-init2.o `test -f 'init2.c' || echo './'`init2.c
mv -f .deps/soapcpp2-init2.Tpo .deps/soapcpp2-init2.Po
gcc -DHAVE_CONFIG_H -I. -I../..    -DWITH_BISON -DWITH_FLEX  -DSOAPCPP2_IMPORT_PATH="\"/usr/local/share/gsoap/import\"" -DLINUX -g -O2 -MT soapcpp2-soapcpp2.o -MD -MP -MF .deps/soapcpp2-soapcpp2.Tpo -c -o soapcpp2-soapcpp2.o `test -f 'soapcpp2.c' || echo './'`soapcpp2.c
mv -f .deps/soapcpp2-soapcpp2.Tpo .deps/soapcpp2-soapcpp2.Po
gcc -DWITH_BISON -DWITH_FLEX  -DSOAPCPP2_IMPORT_PATH="\"/usr/local/share/gsoap/import\"" -DLINUX -g -O2   -o soapcpp2 soapcpp2-soapcpp2_yacc.o soapcpp2-soapcpp2_lex.o soapcpp2-symbol2.o soapcpp2-error2.o soapcpp2-init2.o soapcpp2-soapcpp2.o  -lfl 
/usr/bin/ld: soapcpp2-soapcpp2_lex.o:/home/miky/Code/yi-hack-Allwinner-v2/src/onvif_srvd/onvif_srvd/gsoap-2.8/gsoap/src/soapcpp2_lex.l:53: multiple definition of `yylval'; soapcpp2-soapcpp2_yacc.o:/usr/share/bison++/bison.cc:469: first defined here
collect2: error: ld returned 1 exit status
make[5]: *** [Makefile:414: soapcpp2] Error 1
make[5]: Leaving directory '/home/miky/Code/yi-hack-Allwinner-v2/src/onvif_srvd/onvif_srvd/gsoap-2.8/gsoap/src'
make[4]: *** [Makefile:829: all-recursive] Error 1
make[4]: Leaving directory '/home/miky/Code/yi-hack-Allwinner-v2/src/onvif_srvd/onvif_srvd/gsoap-2.8/gsoap'
make[3]: *** [Makefile:457: all] Error 2
make[3]: Leaving directory '/home/miky/Code/yi-hack-Allwinner-v2/src/onvif_srvd/onvif_srvd/gsoap-2.8/gsoap'
make[2]: *** [Makefile:473: all-recursive] Error 1
make[2]: Leaving directory '/home/miky/Code/yi-hack-Allwinner-v2/src/onvif_srvd/onvif_srvd/gsoap-2.8'
make[1]: *** [Makefile:381: all] Error 2
make[1]: Leaving directory '/home/miky/Code/yi-hack-Allwinner-v2/src/onvif_srvd/onvif_srvd/gsoap-2.8'
./gsoap-2.8/gsoap/wsdl/wsdl2h -d -t ./wsdl/typemap.dat  -o generated/onvif.h  wsdl/devicemgmt.wsdl wsdl/media.wsdl wsdl/ptz.wsdl wsdl/b-2.xsd wsdl/bf-2.xsd wsdl/common.xsd wsdl/onvif.xsd wsdl/t-1.xsd
make: ./gsoap-2.8/gsoap/wsdl/wsdl2h: No such file or directory
make: *** [Makefile:225: generated/onvif.h] Error 127
`

mcanteri, 2020-10-27:
Maybe here you can understand a little bit more where it goes in trouble:

`make[5]: Leaving directory '/home/miky/Code/yi-hack-Allwinner-v2/src/onvif_srvd/onvif_srvd/gsoap-2.8/gsoap'
Making all in src
make[5]: Entering directory '/home/miky/Code/yi-hack-Allwinner-v2/src/onvif_srvd/onvif_srvd/gsoap-2.8/gsoap/src'
/bin/bash ../../ylwrap soapcpp2_yacc.y y.tab.c soapcpp2_yacc.c y.tab.h `echo soapcpp2_yacc.c | sed -e s/cc$/hh/ -e s/cpp$/hpp/ -e s/cxx$/hxx/ -e s/c++$/h++/ -e s/c$/h/` y.output soapcpp2_yacc.output -- bison -y -d -v 
/home/miky/Code/yi-hack-Allwinner-v2/src/onvif_srvd/onvif_srvd/gsoap-2.8/gsoap/src/soapcpp2_yacc.y:157 parser name defined to default :"parse"
conflicts:  1 shift/reduce
updating soapcpp2_yacc.output
updating soapcpp2_yacc.h
gcc -DHAVE_CONFIG_H -I. -I../..    -DWITH_BISON -DWITH_FLEX  -DSOAPCPP2_IMPORT_PATH="\"/usr/local/share/gsoap/import\"" -DLINUX -g -O2 -MT soapcpp2-soapcpp2_yacc.o -MD -MP -MF .deps/soapcpp2-soapcpp2_yacc.Tpo -c -o soapcpp2-soapcpp2_yacc.o `test -f 'soapcpp2_yacc.c' || echo './'`soapcpp2_yacc.c
mv -f .deps/soapcpp2-soapcpp2_yacc.Tpo .deps/soapcpp2-soapcpp2_yacc.Po
/bin/bash ../../ylwrap soapcpp2_lex.l lex.yy.c soapcpp2_lex.c -- flex -l 
gcc -DHAVE_CONFIG_H -I. -I../..    -DWITH_BISON -DWITH_FLEX  -DSOAPCPP2_IMPORT_PATH="\"/usr/local/share/gsoap/import\"" -DLINUX -g -O2 -MT soapcpp2-soapcpp2_lex.o -MD -MP -MF .deps/soapcpp2-soapcpp2_lex.Tpo -c -o soapcpp2-soapcpp2_lex.o `test -f 'soapcpp2_lex.c' || echo './'`soapcpp2_lex.c
In file included from /usr/include/string.h:519,
                 from soapcpp2_lex.c:20:
In function ‘strncpy’,
    inlined from ‘install_pragma’ at soapcpp2_lex.l:332:9,
    inlined from ‘yylex’ at soapcpp2_lex.l:184:10:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:106:10: warning: ‘__builtin_strncpy’ specified bound depends on the length of the source argument [-Wstringop-overflow=]
  106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
soapcpp2_lex.c: In function ‘yylex’:
soapcpp2_lex.l:332:35: note: length computed here
  332 |         strncpy(yylval.s, yytext, strlen(yytext)-1);
      |                                   ^~~~~~~~~~~~~~
mv -f .deps/soapcpp2-soapcpp2_lex.Tpo .deps/soapcpp2-soapcpp2_lex.Po
gcc -DHAVE_CONFIG_H -I. -I../..    -DWITH_BISON -DWITH_FLEX  -DSOAPCPP2_IMPORT_PATH="\"/usr/local/share/gsoap/import\"" -DLINUX -g -O2 -MT soapcpp2-symbol2.o -MD -MP -MF .deps/soapcpp2-symbol2.Tpo -c -o soapcpp2-symbol2.o `test -f 'symbol2.c' || echo './'`symbol2.c
In file included from /usr/include/stdio.h:866,
                 from soapcpp2.h:37,
                 from symbol2.c:37:
In function ‘fprintf’,
    inlined from ‘gen_call_method’ at symbol2.c:8034:5:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:103:10: warning: ‘%s’ directive argument is null [-Wformat-overflow=]
  103 |   return __fprintf_chk (__stream, __USE_FORTIFY_LEVEL - 1, __fmt,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  104 |    __va_arg_pack ());
      |    ~~~~~~~~~~~~~~~~~
/usr/include/x86_64-linux-gnu/bits/stdio2.h:103:10: warning: ‘%s’ directive argument is null [-Wformat-overflow=]
In function ‘fprintf’,
    inlined from ‘gen_call_method’ at symbol2.c:8058:5:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:103:10: warning: ‘%s’ directive argument is null [-Wformat-overflow=]
  103 |   return __fprintf_chk (__stream, __USE_FORTIFY_LEVEL - 1, __fmt,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  104 |    __va_arg_pack ());
      |    ~~~~~~~~~~~~~~~~~
/usr/include/x86_64-linux-gnu/bits/stdio2.h:103:10: warning: ‘%s’ directive argument is null [-Wformat-overflow=]
mv -f .deps/soapcpp2-symbol2.Tpo .deps/soapcpp2-symbol2.Po
gcc -DHAVE_CONFIG_H -I. -I../..    -DWITH_BISON -DWITH_FLEX  -DSOAPCPP2_IMPORT_PATH="\"/usr/local/share/gsoap/import\"" -DLINUX -g -O2 -MT soapcpp2-error2.o -MD -MP -MF .deps/soapcpp2-error2.Tpo -c -o soapcpp2-error2.o `test -f 'error2.c' || echo './'`error2.c
mv -f .deps/soapcpp2-error2.Tpo .deps/soapcpp2-error2.Po
gcc -DHAVE_CONFIG_H -I. -I../..    -DWITH_BISON -DWITH_FLEX  -DSOAPCPP2_IMPORT_PATH="\"/usr/local/share/gsoap/import\"" -DLINUX -g -O2 -MT soapcpp2-init2.o -MD -MP -MF .deps/soapcpp2-init2.Tpo -c -o soapcpp2-init2.o `test -f 'init2.c' || echo './'`init2.c
mv -f .deps/soapcpp2-init2.Tpo .deps/soapcpp2-init2.Po
gcc -DHAVE_CONFIG_H -I. -I../..    -DWITH_BISON -DWITH_FLEX  -DSOAPCPP2_IMPORT_PATH="\"/usr/local/share/gsoap/import\"" -DLINUX -g -O2 -MT soapcpp2-soapcpp2.o -MD -MP -MF .deps/soapcpp2-soapcpp2.Tpo -c -o soapcpp2-soapcpp2.o `test -f 'soapcpp2.c' || echo './'`soapcpp2.c
mv -f .deps/soapcpp2-soapcpp2.Tpo .deps/soapcpp2-soapcpp2.Po
gcc -DWITH_BISON -DWITH_FLEX  -DSOAPCPP2_IMPORT_PATH="\"/usr/local/share/gsoap/import\"" -DLINUX -g -O2   -o soapcpp2 soapcpp2-soapcpp2_yacc.o soapcpp2-soapcpp2_lex.o soapcpp2-symbol2.o soapcpp2-error2.o soapcpp2-init2.o soapcpp2-soapcpp2.o  -lfl 
/usr/bin/ld: soapcpp2-soapcpp2_lex.o:/home/miky/Code/yi-hack-Allwinner-v2/src/onvif_srvd/onvif_srvd/gsoap-2.8/gsoap/src/soapcpp2_lex.l:53: multiple definition of `yylval'; soapcpp2-soapcpp2_yacc.o:/usr/share/bison++/bison.cc:469: first defined here
collect2: error: ld returned 1 exit status
make[5]: *** [Makefile:414: soapcpp2] Error 1
make[5]: Leaving directory '/home/miky/Code/yi-hack-Allwinner-v2/src/onvif_srvd/onvif_srvd/gsoap-2.8/gsoap/src'
[... cut ...]
make[2]: Leaving directory '/home/miky/Code/yi-hack-Allwinner-v2/src/onvif_srvd/onvif_srvd/gsoap-2.8'
make[1]: *** [Makefile:381: all] Error 2
make[1]: Leaving directory '/home/miky/Code/yi-hack-Allwinner-v2/src/onvif_srvd/onvif_srvd/gsoap-2.8'
./gsoap-2.8/gsoap/wsdl/wsdl2h -d -t ./wsdl/typemap.dat  -o generated/onvif.h  wsdl/devicemgmt.wsdl wsdl/media.wsdl wsdl/ptz.wsdl wsdl/b-2.xsd wsdl/bf-2.xsd wsdl/common.xsd wsdl/onvif.xsd wsdl/t-1.xsd
make: ./gsoap-2.8/gsoap/wsdl/wsdl2h: No such file or directory
make: *** [Makefile:225: generated/onvif.h] Error 127`

roleoroleo, 2020-10-27:
Did you update with this commit?
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/d553ac3b5e8077459cc48217b592181348256b87

mcanteri, 2020-10-27:
> Did you update with this commit?
> [d553ac3](https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/d553ac3b5e8077459cc48217b592181348256b87)

Yes, I made a git pull and I verified now, I got that.

roleoroleo, 2020-10-28:
I totally rebuilt the repo and it's working.
Maybe some issues with bison/flex (host side)?

mcanteri, 2020-10-28:
Yes, could be, entered in the problematic directory and run make clean, make all, I have:

mv -f .deps/soapcpp2-soapcpp2.Tpo .deps/soapcpp2-soapcpp2.Po
gcc -DWITH_BISON -DWITH_FLEX  -DSOAPCPP2_IMPORT_PATH="\"/usr/local/share/gsoap/import\"" -DLINUX -g -O2   -o soapcpp2 soapcpp2-soapcpp2_yacc.o soapcpp2-soapcpp2_lex.o soapcpp2-symbol2.o soapcpp2-error2.o soapcpp2-init2.o soapcpp2-soapcpp2.o  -lfl 
/usr/bin/ld: soapcpp2-soapcpp2_lex.o:/home/miky/Code/yi-hack-Allwinner-v2/src/onvif_srvd/onvif_srvd/gsoap-2.8/gsoap/src/soapcpp2_lex.l:53: multiple definition of `yylval'; soapcpp2-soapcpp2_yacc.o:/usr/share/bison++/bison.cc:469: first defined here
collect2: error: ld returned 1 exit status
make: *** [Makefile:414: soapcpp2] Error 1

It looks bison but I don't know how to fix it.

roleoroleo, 2020-10-28:
Try -fno-common
https://sourceforge.net/p/gsoap2/bugs/1269/

mcanteri, 2020-10-28:
> Try -fno-common
> https://sourceforge.net/p/gsoap2/bugs/1269/

Tried all possible about that but isn't solving the problem.

mcanteri, 2020-10-28:
> > Try -fno-common
> > https://sourceforge.net/p/gsoap2/bugs/1269/
> 
> Tried all possible about that but isn't solving the problem.

Last try was to delete all and start from scratch, edited the file soapcpp2_lex.l,  added extern to the variable, then ./configure CFLAGS="-O0 -ggdb3 -fno-common" and make : this solve the compilation in that folder, but then when I execute again ./script/compile it goes again in the problem. Maybe something is deleted before the next compile.
If you try that workaround will compile on your env ?

c3kkos, 2020-11-02:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases

this page hold the release, current 0.1.2

roleoroleo, 2020-11-02:
@mcanteri 
I'm trying to build a newer version of gsoap but I have a lot of problems.

mcanteri, 2020-11-02:
I think it is difficult that I can be of any help there, but if you imagine something, let me know :+1:

closed 2022-01-20 by roleoroleo