roleoroleo/yi-hack-Allwinner-v2 #10: High res with r30gb

roleoroleo, 2020-10-27:
@artemy 
I understand why high resolution doesn't work: the high res stream is h265 coded.
The old models (y21ga too) use h264 for both, high res and low res stream.
r30gb uses h264 for low res stream and h265 for high res stream.
I will try to solve the problem, but is not simple.

roleoroleo, 2020-10-27:
Try this binary:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5446733/rRTSPServer.gz)

channel0 is an h265 stream.
At the moment the address is rtsp://IP-CAM/ch0_0.h264

artemy, 2020-10-27:
VLC can't open the high-res stream. Low-res works fine still.

roleoroleo, 2020-10-27:
Please, another try:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5447664/rRTSPServer.gz)

artemy, 2020-10-27:
Both streams do seem to work alright now.

wimpie007, 2020-10-27:
@artemy highres confirmed??

roleoroleo, 2020-10-27:
Please try snapshot too:
[snapshot.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5447784/snapshot.tar.gz)

artemy, 2020-10-27:
> Please try snapshot too:
> [snapshot.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5447784/snapshot.tar.gz)

Both are broken. Low-res just shows green screen, high-res doesn't give any output at all

roleoroleo, 2020-10-27:
Another try?
:)
[snapshot.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5447903/snapshot.tar.gz)

artemy, 2020-10-28:
So both are good now, although I think there is another issue now. 
I can get a high-res snapshot once after camera boots, but then it randomly causes camera reboot after one of the next requests

roleoroleo, 2020-10-28:
The problem here is the memory.
The cam runs with very little ram available and ffmpeg library is very "big".
Try to enable swap space.

closed 2020-12-27 by roleoroleo