roleoroleo/yi-hack-Allwinner-v2 #11: RTSP not working using FFmpeg

fklement, 2020-10-27:
Somehow i can't get `RSTP` working with FFmpeg.

See debug log down below:
```
[10/27/2020, 13:08:36] [Camera FFmpeg] [Küche] Snapshot requested: 480 x 270
[10/27/2020, 13:08:36] [Camera FFmpeg] [Küche] Sending snapshot: 480 x 270
[10/27/2020, 13:08:36] [Camera FFmpeg] [Küche] Snapshot command: /usr/lib/node_modules/homebridge-camera-ffmpeg/node_modules/ffmpeg-for-homebridge/ffmpeg -i http://192.168.1.122:8080/cgi-bin/snapshot.sh  -frames:v 1 -filter:v scale='min(480,iw)':'min(270,ih)':force_original_aspect_ratio=decrease,scale=trunc(iw/2)*2:trunc(ih/2)*2 -f image2 -
[10/27/2020, 13:08:43] [Camera FFmpeg] [Küche] Video stream requested: 1280 x 720, 30 fps, 299 kbps
[10/27/2020, 13:08:43] [Camera FFmpeg] [Küche] Starting video stream: 1280 x 720, 30 fps, 299 kbps
[10/27/2020, 13:08:43] [Camera FFmpeg] [Küche] Stream command: /usr/lib/node_modules/homebridge-camera-ffmpeg/node_modules/ffmpeg-for-homebridge/ffmpeg -i rtsp://cam1:batavis@192.168.1.122:554/ch0_0.h264 -an -sn -dn -codec:v libx264 -pix_fmt yuv420p -color_range mpeg -r 30 -f rawvideo -preset ultrafast -tune zerolatency -filter:v scale='min(1280,iw)':'min(720,ih)':force_original_aspect_ratio=decrease,scale=trunc(iw/2)*2:trunc(ih/2)*2 -b:v 299k -payload_type 99 -ssrc 8556460 -f rtp -srtp_out_suite AES_CM_128_HMAC_SHA1_80 -srtp_out_params 9PhXuPPuZG5WlOSEpTygkZKJQkeGbfTCrvg88jN8 srtp://192.168.178.202:62725?rtcpport=62725&pkt_size=1316 -vn -sn -dn -codec:a libfdk_aac -profile:a aac_eld -flags +global_header -f null -ar 16k -b:a 24k -ac 1 -payload_type 110 -ssrc 5651798 -f rtp -srtp_out_suite AES_CM_128_HMAC_SHA1_80 -srtp_out_params lm43GneVopeov3qlFiZ1synlnFdM1cYE33IA56cz srtp://192.168.178.202:53131?rtcpport=53131&pkt_size=188 -loglevel level+verbose
[10/27/2020, 13:08:43] [Camera FFmpeg] [Küche] [info] ffmpeg version 4.3.1 Copyright (c) 2000-2020 the FFmpeg developers
[10/27/2020, 13:08:43] [Camera FFmpeg] [Küche] [info]   built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1+deb9u1) 20170516
[10/27/2020, 13:08:43] [Camera FFmpeg] [Küche] [info]   configuration: --pkgconfigdir=/build/workspace/lib/pkgconfig --prefix=/build/workspace --pkg-config-flags=--static --extra-cflags=-I/build/workspace/include --extra-ldflags=-L/build/workspace/lib --extra-libs='-lpthread -lm' --enable-static --disable-debug --disable-shared --disable-ffplay --disable-doc --enable-openssl --enable-gpl --enable-version3 --enable-nonfree --enable-pthreads --enable-libvpx --enable-libmp3lame --enable-libopus --enable-libtheora --enable-libvorbis --enable-libx264 --enable-runtime-cpudetect --enable-libfdk-aac --enable-avfilter --enable-libopencore_amrwb --enable-libopencore_amrnb --enable-filters --enable-decoder=h264 --enable-network --enable-protocol=tcp --enable-libspeex --enable-demuxer=rtsp --enable-omx-rpi --enable-mmal --enable-indev=alsa --enable-outdev=alsa
[10/27/2020, 13:08:43] [Camera FFmpeg] [Küche] 
[10/27/2020, 13:08:43] [Camera FFmpeg] [Küche] [info]   libavutil      56. 51.100 / 56. 51.100
[10/27/2020, 13:08:43] [Camera FFmpeg] [Küche] [info]   libavcodec     58. 91.100 / 58. 91.100
[10/27/2020, 13:08:43] [Camera FFmpeg] [Küche] [info]   libavformat    58. 45.100 / 58. 45.100
[10/27/2020, 13:08:43] [Camera FFmpeg] [Küche] [info]   libavdevice    58. 10.100 / 58. 10.100
[10/27/2020, 13:08:43] [Camera FFmpeg] [Küche] [info]   libavfilter     7. 85.100 /  7. 85.100
[10/27/2020, 13:08:43] [Camera FFmpeg] [Küche] [info]   libswscale      5.  7.100 /  5.  7.100
[10/27/2020, 13:08:43] [Camera FFmpeg] [Küche] [info]   libswresample   3.  7.100 /  3.  7.100
[10/27/2020, 13:08:43] [Camera FFmpeg] [Küche] [info]   libpostproc    55.  7.100 / 55.  7.100
[10/27/2020, 13:08:43] [Camera FFmpeg] [Küche] 
[10/27/2020, 13:08:43] [Camera FFmpeg] [Küche] [tcp @ 0x3f79a20] [verbose] Starting connection attempt to 192.168.1.122 port 554
[10/27/2020, 13:08:43] [Camera FFmpeg] [Küche] 
[10/27/2020, 13:08:43] [Camera FFmpeg] [Küche] [tcp @ 0x3f79a20] [verbose] Successfully connected to 192.168.1.122 port 554
[10/27/2020, 13:08:43] [Camera FFmpeg] [Küche] 
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] [rtsp @ 0x3f779f0] [verbose] SDP:
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] v=0
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] o=- 1603804032870323 1 IN IP4 192.168.1.122
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] s=Session streamed by "rRTSPServer"
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] i=ch0_0.h264
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] t=0 0
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] a=tool:LIVE555 Streaming Media v2020.05.15
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] a=type:broadcast
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] a=control:*
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] a=range:npt=0-
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] a=x-qt-text-nam:Session streamed by "rRTSPServer"
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] a=x-qt-text-inf:ch0_0.h264
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] m=video 0 RTP/AVP 96
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] c=IN IP4 0.0.0.0
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] b=AS:500
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] a=rtpmap:96 H264/90000
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] a=fmtp:96 packetization-mode=1;profile-level-id=4D0020;sprop-parameter-sets=Z00AIJZUA8ARLyzcBAQECA==,aO48gA==
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] a=control:track1
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] m=audio 0 RTP/AVP 0
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] c=IN IP4 0.0.0.0
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] b=AS:64
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] a=control:track2
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] 
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] 
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] [udp @ 0x3f7c910] [warning] attempted to set receive buffer to size 393216 but it only ended up set as 360448
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] [udp @ 0x3f8cc70] [warning] attempted to set receive buffer to size 393216 but it only ended up set as 360448
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] [rtsp @ 0x3f779f0] [verbose] setting jitter buffer size to 500
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] [udp @ 0x3f9d9e0] [warning] attempted to set receive buffer to size 393216 but it only ended up set as 360448
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] [udp @ 0x3fadd30] [warning] attempted to set receive buffer to size 393216 but it only ended up set as 360448
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] 
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] [rtsp @ 0x3f779f0] [verbose] setting jitter buffer size to 500
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] 
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] [h264 @ 0x3f7ae90] [verbose] Reinit context to 1920x1088, pix_fmt: yuvj420p
[10/27/2020, 13:08:45] [Camera FFmpeg] [Küche] 
[10/27/2020, 13:08:48] [Camera FFmpeg] [Küche] [warning] Guessed Channel Layout for Input Stream #0.1 : mono
[10/27/2020, 13:08:48] [Camera FFmpeg] [Küche] [info] Input #0, rtsp, from 'rtsp://cam1:batavis@192.168.1.122:554/ch0_0.h264':
[10/27/2020, 13:08:48] [Camera FFmpeg] [Küche] 
[10/27/2020, 13:08:48] [Camera FFmpeg] [Küche] [info]   Metadata:
[10/27/2020, 13:08:48] [Camera FFmpeg] [Küche] [info]     title           : Session streamed by "rRTSPServer"
[10/27/2020, 13:08:48] [Camera FFmpeg] [Küche] [info]     comment         : ch0_0.h264
[10/27/2020, 13:08:48] [Camera FFmpeg] [Küche] [info]   Duration: N/A, start: 0.000000, bitrate: N/A
[10/27/2020, 13:08:48] [Camera FFmpeg] [Küche] [info]     Stream #0:0: Video: h264 (Main), 1 reference frame, yuvj420p(pc, bt709, progressive, left), 1920x1080 (1920x1088), 20 fps, 20 tbr, 90k tbn, 180k tbc
[10/27/2020, 13:08:48] [Camera FFmpeg] [Küche] [info]     Stream #0:1: Audio: pcm_mulaw, 8000 Hz, mono, s16, 64 kb/s
[... 168 lines cut ...]
[10/27/2020, 13:09:08] [Camera FFmpeg] [Küche] [verbose] *** 1 dup!
[10/27/2020, 13:09:08] [Camera FFmpeg] [Küche] 
[10/27/2020, 13:09:08] [Camera FFmpeg] [Küche] [info] frame=  151 fps=7.6 q=21.0 size=     125kB time=00:00:05.00 bitrate= 204.4kbits/s dup=78 drop=0 speed=0.251x    
[10/27/2020, 13:09:08] [Camera FFmpeg] [Küche] [rtsp @ 0x3f779f0] [warning] max delay reached. need to consume packet
[10/27/2020, 13:09:08] [Camera FFmpeg] [Küche] [rtsp @ 0x3f779f0] [warning] RTP: missed 34 packets
[10/27/2020, 13:09:08] [Camera FFmpeg] [Küche] 
[10/27/2020, 13:09:08] [Camera FFmpeg] [Küche] [h264 @ 0x40de010] 
[... cut ...]
[10/27/2020, 13:09:08] [Camera FFmpeg] [Küche] 
[10/27/2020, 13:09:08] [Camera FFmpeg] [Küche] [h264 @ 0x408a220] [info] concealing 409 DC, 409 AC, 409 MV errors in I frame
[10/27/2020, 13:09:08] [Camera FFmpeg] [Küche] 
[10/27/2020, 13:09:09] [Camera FFmpeg] [Küche] [verbose] *** 305 dup!
[10/27/2020, 13:09:09] [Camera FFmpeg] [Küche] 
[10/27/2020, 13:09:13] [Camera FFmpeg] [Küche] Stopped video stream.
[10/27/2020, 13:09:13] [Camera FFmpeg] [Küche] FFmpeg exited with code: null and signal: SIGKILL (Expected)
```

mcanteri, 2020-10-27:
Please remove your passwd from log 👍

fklement, 2020-10-27:
@mcanteri I use this camera currently only for testing and not in production. So no problem here.

mcanteri, 2020-10-27:
Never a good idea to share those information, but as you wish.

roleoroleo, 2020-10-27:
Please, check this issue:
https://github.com/roleoroleo/yi-hack-MStar/issues/36

closed 2020-12-27 by roleoroleo