roleoroleo/yi-hack-Allwinner-v2 #12: No compiled Firmware Files in Release

mapan-de, 2020-10-27:
Hello,
please check.

I can't find the Firmware Files in the downloadable release. I also tried to compile it myself on a Linux-PC, but it stopped with errors.

Thanks in advance!

roleoroleo, 2020-10-27:
Currently this project is in beta release.
No official release.

fklement, 2020-10-27:
See this Issue:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/8

> Sorry, you can find beta releases in this post:
> [#5 (comment)](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/5#issuecomment-716736627)
> or in this issue:
> #3

damanti-me, 2020-10-27:
@mapan-de you can use 0.0.8 for now. RTSP stream without audio and many other things are working for testing.

mapan-de, 2020-10-27:
Thanks for the fast response!
The Cam I am using is a -> Yi 1080p Home BFUS | 9.0.19* | y21ga <-
If that is already supported in 0.0.8 I would be happy to know a link for that download - I am not so comfort with github and couldn't find it in releases.

Thanks in advance, again!

mcanteri, 2020-10-27:
@mapan-de now a release is present, you can download it.

closed 2020-10-28 by roleoroleo