roleoroleo/yi-hack-Allwinner-v2 #13: y21ga 0.1.2 issues and new tests

mcanteri, 2020-10-27:
- The command gzip is not working, but this even in previous version, I tried to launch it and it says:
gzip: applet not found

- Streams miss audio, video works both with direct player (i.e. mplayer) and integrated with HA. I hope it'll come back because if I'm not wrong at the beginning it was working.

MQTT sensor works integrated with HA.
Looks quite stable, good release! :) :+1:

mcanteri, 2020-10-27:
Forgot, I used the upgrade firmware feature on the Maintenance page and it worked :+1:

pep4n, 2020-10-27:
> 
> 
> Forgot, I used the upgrade firmware feature on the Maintenance page and it worked 👍

coming from 0.1.1?

mcanteri, 2020-10-27:
> coming from 0.1.1?

I was on the 0.1.0 without boot loop (partial upgrade, only the folder yi-hack).

> In general do you guys activate the sd swap feature?

Yes

pep4n, 2020-10-27:
cool - so even from macgyver version OTA update to 0.1.2 ,) 
this hack is getting pretty pro 👍 

let me repeat at this point: fantastic job remote reverse engineering this cam and thank you so much @roleoroleo !

mcanteri, 2020-10-27:
I agree, it runs very nice, smoothly with Home Assistant too!
Great job @roleoroleo !

LukasVFL99, 2020-10-28:
Works great 😊 I made the update with ota. Thanks @roleoroleo

pep4n, 2020-10-28:
Guys please lets test the following on 0.1.2 and give feedback here:  

    rtsp high res - yes but with more artefacts in low light than before. gets worse the longer the stream runs. 
    rtsp low res - not tested yet 
    rtsp audio - no
    snapshot high res - yes
    snapshot low res  yes
    status page - yes
    hostname - updates
    timezone - works after giving the string
    onvif - not tested yet
    ftp upload - not tested yet
    mp4 management in "Motion Events" page - not tested completely but seems to work 
    web snapshot - yes
    ptz (only R30 model) - n/a
    load/save configuration - save yes. load tbd.

roleoroleo, 2020-10-29:
@pep4n 
Try to run rRTSPServer manually with debug option.
Check if there are errors.

pep4n, 2020-10-29:
Ok, I guess I would need some help running this. 
Connect via SSH or telnet?

roleoroleo, 2020-10-29:
Connect via SSH:
```
. /tmp/sd/yi-hack/script/env.sh
killall rRTSPServer
rRTSPServer -r both -d
```

mcanteri, 2020-10-29:
Another minor issue is that the check about free space on SD card is not working.
I set up for 30% free space and now it is at 15%.

mcanteri, 2020-10-29:
> rtsp high res - tested via VLC: yes but with more artefacts in low light than before. gets worse the longer the stream runs.  see screenshot attached. also probably around 2fps. need to compare to daylight though. 

Mine is much better on low light that the quality of image you posted.

pep4n, 2020-10-29:
> 
> 
> Connect via SSH:
> 
> ```
> . /tmp/sd/yi-hack/script/env.sh
> killall rRTSPServer
> rRTSPServer -r both -d
> ```

putty can't connect: access denied. 
tried with set user and password and also removing credentials doesnt help...

roleoroleo, 2020-10-29:
> Failed to create RTSP server: bind() error (port number: 554): Address in use

After killing rRTSPServer you have to wait 20-30 seconds, the system doesn't release the socket immediately.

mcanteri, 2020-10-29:
> > Failed to create RTSP server: bind() error (port number: 554): Address in use
> 
> After killing rRTSPServer you have to wait 20-30 seconds, the system doesn't release the socket immediately.

There is the watchdog that restart the server, so if you wait it'll start again.
Try this:

. /tmp/sd/yi-hack/script/env.sh
ps| grep rRTSPServer | grep -v grep | cut -d ' ' -f3 | xargs kill ; rRTSPServer -r both -d 

You should have a queue of:
frame_len: 750 - cb_current->size: 49152
frame_len: 1530 - cb_current->size: 196608

To interrupt use Ctrl+C

roleoroleo, 2020-10-29:
@pep4n 
Try this binary:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5458814/rRTSPServer.gz)
I increased verbosity.

roleoroleo, 2020-10-29:
> There is the watchdog that restart the server, so if you wait it'll start again.

Kill wd_rtsp.sh too.

mcanteri, 2020-10-29:
> > There is the watchdog that restart the server, so if you wait it'll start again.
> 
> Kill wd_rtsp.sh too.

If you don't kill the watchdog, the RSTP server will restart again without need to reboot or manually start the watchdog :)
One tool that would be useful in this case would be 'tee' , to see and log the output. Tried with busybox but it isn't included.

pep4n, 2020-10-29:
#

> 
> 
> @pep4n
> Try this binary:
> [rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5458814/rRTSPServer.gz)
> I increased verbosity.

again sorry guys, but how can I test this? 
:)

mcanteri, 2020-10-29:
> > @pep4n
> > Try this binary:
> > [rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5458814/rRTSPServer.gz)
> > I increased verbosity.
> 
> again sorry guys, but how can I test this?
> :)

Are you familiar with Linux or Windows ?

pep4n, 2020-10-29:
just a little little bit and yes a lot. 
guess what goes for linux ,)

roleoroleo, 2020-10-29:
Copy the file using ftp or scp client to /tmp/sd/yi-hack/bin (Filezilla or WinSCP are ok).
Connect with putty (user root without password) and run the following commands:
```
. /tmp/sd/yi-hack/script/env.sh
killall rRTSPServer
cd /tmp/sd/yi-hack/bin
mv rRTSPServer rRTSPServer.ori
gzip -d rRTSPServer.gz
rRTSPServer -r both -d
```
The 1st command load the envornment variables for the current shell.
The 2nd command terminates the process rRTSPServer.
The last command run the new rRTSPServer.

pep4n, 2020-10-29:
awesome - thanks. will do and return with feeback. 
or failure ,)

pep4n, 2020-10-30:
I could update the rRTSPServer. 
Now I should do the "rRTSPServer -r both -d" again and post output, right?

mcanteri, 2020-10-30:
So you moved the old binary and substituted it with the new one, then you can run this line:

ps| grep rRTSPServer | grep -v grep | cut -d ' ' -f3 | xargs kill ; rRTSPServer -r both -d

and see what is the output.

If you want to save the output, run this:

ps| grep rRTSPServer | grep -v grep | cut -d ' ' -f3 | xargs kill ; rRTSPServer -r both -d 2> rtsp.log

and share the file rtsp.log

pep4n, 2020-10-30:
I get the following when trying to run this: 
xargs: not found

mcanteri, 2020-10-31:
> I get the following when trying to run this:
> xargs: not found

Do:

cp /tmp/sd/yi-hack/bin/gzip /tmp/sd/yi-hack/bin/xargs

then try again.

pep4n, 2020-10-31:
hmm still. 
```

root@yi-cam:~# cp /tmp/sd/yi-hack/bin/gzip /tmp/sd/yi-hack/bin/xargs

root@yi-cam:~# ps| grep rRTSPServer | grep -v grep | cut -d ' '-f3 | xargs kill ; rRTSPServer -r both -d 2> rtsp.log
-ash: xargs: not found

root@yi-cam:~#
```

mcanteri, 2020-10-31:
Have you:
. /tmp/sd/yi-hack/script/env.sh

If still not working, reboot the cam then it'll work.

pep4n, 2020-10-31:
aha seems we're one step further: 
`kill: you need to specify whom to kill`

I like this one :D

roleoroleo, 2020-10-31:
Remember to source the env.
First line:

> ```
> . /tmp/sd/yi-hack/script/env.sh
> killall rRTSPServer
> cd /tmp/sd/yi-hack/bin
> mv rRTSPServer rRTSPServer.ori
> gzip -d rRTSPServer.gz
> rRTSPServer -r both -d
> ```

pep4n, 2020-10-31:
I run this line by line right?

roleoroleo, 2020-10-31:
Only env.

roleoroleo, 2020-10-31:
Ok try this:
```
. /tmp/sd/yi-hack/script/env.sh
killall wd_rtsp.sh
killall rRTSPServer
cd /tmp/sd/yi-hack/bin
mv rRTSPServer rRTSPServer.ori
gzip -d rRTSPServer.gz
rRTSPServer -r both -d
```

mcanteri, 2020-10-31:
> Ok try this:
> 
> ```
> . /tmp/sd/yi-hack/script/env.sh
> killall wd_rtsp.sh
> killall rRTSPServer
> cd /tmp/sd/yi-hack/bin
> mv rRTSPServer rRTSPServer.ori
> gzip -d rRTSPServer.gz
> rRTSPServer -r both -d
> ```

We will never see the light ! :-D GZIP isn't working but okay, maybe @pep4n can transfer it uncompressed :+1:

mcanteri, 2020-10-31:
I improved the command, try this:

`ps | grep rRTSPServer | grep -v grep | awk '{print $1}' | xargs kill ; rRTSPServer -r both -d 2> rtsp.log`

This time should work.

mcanteri, 2020-10-31:
Another minor issue found, we miss the sort command so the ftp feature cannot work: I have some ftppush.sh hanging.
So, till now, have to fix gzip and sort.

roleoroleo, 2020-11-01:
Ok.

mcanteri, 2020-11-01:
The rtsp server isn't perfect, today I enabled my automation that record a video when something happen (now camera motion mqtt sensor) using the camera.record feature of HA and sometime the video is readable but sometime it isn't.
When it is not readable I found this exception in the HA log:
`2020-11-01 17:40:37 ERROR (recorder_save_worker) [libav.mp4] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 13434267 >= -37381920
2020-11-01 17:40:37 ERROR (recorder_save_worker) [root] Uncaught thread exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/src/homeassistant/homeassistant/components/stream/recorder.py", line 51, in recorder_save_worker
    output.mux(packet)
  File "av/container/output.pyx", line 204, in av.container.output.OutputContainer.mux
  File "av/container/output.pyx", line 227, in av.container.output.OutputContainer.mux_one
  File "av/container/core.pyx", line 257, in av.container.core.Container.err_check
  File "av/error.pyx", line 336, in av.error.err_check
av.error.ValueError: [Errno 22] Invalid argument: '/media/surveillance/LivingRoom_camera.mp4'; last error log: [mp4] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 13434267 >= -37381920
`

roleoroleo, 2020-11-01:
> Another minor issue found, we miss the sort command so the ftp feature cannot work: I have some ftppush.sh hanging.
> So, till now, have to fix gzip and sort.

https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/5fe19b4113e57e8f07440f8034efa11fb184244d

mcanteri, 2020-11-02:
For FTP push to work, missing nc and ftpput too.
Then a small change (look at the diff) to the ftppush.sh script, starting at row 159:

`       if [ "${FTP_DIR_TREE}" == "yes" ] ; then                                                                                                                        
                if ( ! ftpput -u "${FTP_USERNAME}" -p "${FTP_PASSWORD}" "${FTP_HOST}" "${FTP_DIR}${FTP_DIR_HOUR}$(lbasename "${UTF_FULLFN}")" "${UTF_FULLFN}" ); then   
                        logAdd "[ERROR] uploadToFtp: ftpput FAILED."                                                                                                    
                        return 1                                                                                                                                        
                fi                                                                                                                                                      
        else                                                                                                                                                            
                if ( ! ftpput -u "${FTP_USERNAME}" -p "${FTP_PASSWORD}" "${FTP_HOST}" "${FTP_DIR}$(lbasename "${UTF_FULLFN}")" "${UTF_FULLFN}" ); then                  
                        logAdd "[ERROR] uploadToFtp: ftpput FAILED."                                                                                                    
                        return 1                                                                                                                                        
                fi                                                                                                                                                      
        fi                                                                                                                                                              
`
A / was breaking the ftpput. Replaced echo with logAdd.

roleoroleo, 2020-11-03:
Done.
Thank you.

closed 2020-12-27 by roleoroleo