roleoroleo/yi-hack-Allwinner-v2 #9: Snapshot not working - V.0.1.1

fklement, 2020-10-27:
Somehow I only get a green picture.
But I have not yet done any further investigation.

[image]

RTSP Stream is working just fine.

marcomehnert, 2020-10-27:
Same here. But I am happy to get that far to have a partly working hack for the new version of the cam. OTA-Update did not work. It shows an older version than the one that is installed.

mcanteri, 2020-10-27:
Which version of the hack are you referring (r30gb or y21ga) ?

LukasVFL99, 2020-10-27:
y21ga @mcanteri

fklement, 2020-10-27:
@mcanteri I'm also using y21ga

mcanteri, 2020-10-27:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/ae8d3932c96e9fbe5c72372d8c5f237b7461a1d1

Repository says it is still 0.1.0 , where did you get it ? :)

marcomehnert, 2020-10-27:
a zipped package with version 0.1.1 was linked somewhere in the issue comments

marcomehnert, 2020-10-27:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/5#issuecomment-716736627

mcanteri, 2020-10-27:
Ok, thank you, it's still in test for that issue.

LukasVFL99, 2020-10-27:
Where to get this version? @mcanteri

mcanteri, 2020-10-27:
@LukasVFL99 it was available in the other thread but seems now is no more, so my comment was for those that is following this hack from early days.
Probably in that 0.1.1 Roleo fixed the problem with boot loop if you haven't experienced, so it's good to stay with it too 👍

roleoroleo, 2020-10-27:
Is low res snapshot working?

pep4n, 2020-10-27:
> 
> 
> Is low res snapshot working?

no. but not like shown above. 
For me its not even loading. showing an error message something like this: 
"the picture [image] can not be loaded, because it contains errors."

artemy, 2020-10-27:
I have somewhat similar picture with high-res snapshot on r30gb.

roleoroleo, 2020-10-27:
Please try this binary:
[framefinder.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5445232/framefinder.gz)

mcanteri, 2020-10-27:
> Please try this binary:
> [framefinder.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5445232/framefinder.gz)

All snapshots are working now on y21ga.

fklement, 2020-10-27:
> > Please try this binary:
> > [framefinder.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5445232/framefinder.gz)
> 
> All snapshots are working now on y21ga.

Yes I also can confirm this! 👍🏼

roleoroleo, 2020-10-27:
Ok, I will try to release a new official release.
I hope it turns out better than 0.1.0 :)

mcanteri, 2020-10-27:
For sure it'll be better :-) 👍

roleoroleo, 2020-10-27:
Done!

LukasVFL99, 2020-10-27:
Where to find @roleoroleo ?

patschi1983, 2020-10-27:
> Where to find @roleoroleo ?

In the Release-Section ;-)

closed 2020-10-28 by roleoroleo