roleoroleo/yi-hack-Allwinner-v2 #14: BFUSY44GPV firm 9.0.19.02_202007141512

Sildenafil99, 2020-10-28:
Hi,
today I got these cameras but the hack doesn't work.
Any help?

Sildenafil99, 2020-10-28:
I have no yellow led flashing but fixed...

mcanteri, 2020-10-28:
Are you using the original cable and wall adaptor ?

Sildenafil99, 2020-10-28:
With the original cable and wall adaptor, i have first yellow led fix, after two seconds blue led flashing and then the camera is ready but with the original firmware.
I have extract the y21ga_0.1.2 into the root of the sd card ( it contain 3 folder Factory, newbackup, yi-hack and 1 file lower_half_init.sh) is it correct?
Thx

pep4n, 2020-10-28:
> I have extract the y21ga_0.1.2 into the root of the sd card ( it contain 3 folder Factory, newbackup, yi-hack and 1 file lower_half_init.sh) is it correct?

that's correct.

Sildenafil99, 2020-10-28:
> > I have extract the y21ga_0.1.2 into the root of the sd card ( it contain 3 folder Factory, newbackup, yi-hack and 1 file lower_half_init.sh) is it correct?
> 
> that's correct.

Why the hack don't work?

damanti-me, 2020-10-28:
Is your SD in FAT32 format?

Sildenafil99, 2020-10-29:
> Is your SD in FAT32 format?

Yes, I tried with two different sd card, I tried to format via windows FAT32/4096 and via Yi camera and i have the same result...

Sildenafil99, 2020-10-29:
Help?

pep4n, 2020-10-29:
It seems your cam is not supported. 

You gave the DID BFUSY44GPV
But what is the SN? It should start similar to this: by21ga ...

Sildenafil99, 2020-10-29:
Yes the sn Is by21gaeuw.....

roleoroleo, 2020-10-29:
Try the backup procedure in the wiki.
Check if it works.

Sildenafil99, 2020-10-29:
This is the backup
[backup.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5458351/backup.zip)

mcanteri, 2020-10-29:
> > Is your SD in FAT32 format?
> 
> Yes, I tried with two different sd card, I tried to format via windows FAT32/4096 and via Yi camera and i have the same result...

What is the size of the SD cards you are trying ? Do you have a small one, like 4GB or 8GB to try ?

mcanteri, 2020-10-29:
> This is the backup
> [backup.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5458351/backup.zip)

It is exactly as ours, so it must be something else.

Sildenafil99, 2020-10-29:
> > > Is your SD in FAT32 format?
> > 
> > 
> > Yes, I tried with two different sd card, I tried to format via windows FAT32/4096 and via Yi camera and i have the same result...
> 
> What is the size of the SD cards you are trying ? Do you have a small one, like 4GB or 8GB to try ?

16gb... i don't have 8 or 4 sd card, i can try to create a 4gb partition...

pep4n, 2020-10-29:
fyi I'm using 16gb sandisk as well.

Sildenafil99, 2020-10-29:
I can try to reset cam and choise a different server?

Sildenafil99, 2020-10-29:
I have reset the camera and chose a different localization on tha app, set the camera and the power off and insert the sd but nothing change,  yellow led fix, after two seconds blue led flashing and then the camera is ready but with the original firmware.

roleoroleo, 2020-10-29:
The backup procedure uses the same hack.
And I can confirm that your cam is a y21ga.
It must work.

mcanteri, 2020-10-29:
> I have reset the camera and chose a different localization on tha app, set the camera and the power off and insert the sd but nothing change, yellow led fix, after two seconds blue led flashing and then the camera is ready but with the original firmware.

Sorry, but when the cam is online have you tried to connect using ssh ? (ssh root@YOUR_LAN-IP)

Sildenafil99, 2020-10-29:
> > I have reset the camera and chose a different localization on tha app, set the camera and the power off and insert the sd but nothing change, yellow led fix, after two seconds blue led flashing and then the camera is ready but with the original firmware.
> 
> Sorry, but when the cam is online have you tried to connect using telnet ?

No i have tried this with chrome:
Access the web interface by entering the following in your web browser: http://yi-hack-v4

Depending upon your network setup, accessing the web interface with the hostname may not work. In this case, the IP address of the camera has to be found.

This can be done from the App. Open it and go to the Camera Settings --> Network Info --> IP Address.

Access the web interface by entering the IP address of the came in a web browser. e.g. http://192.168.1.5

mcanteri, 2020-10-29:
> Sorry, but when the cam is online have you tried to connect using telnet ?

Try open http://yourLanIp:8080  with Chrome

mcanteri, 2020-10-29:
> > Sorry, but when the cam is online have you tried to connect using telnet ?
> 
> Try open http://yourLanIp:8080 with Chrome

Ok, accordingly to what you wrote, try to open in Chrome http://192.168.1.5:8080

Sildenafil99, 2020-10-29:
Ok, i'm feel so stupid now... I forgot to writer the port...

closed 2020-10-29 by Sildenafil99