roleoroleo/yi-hack-Allwinner-v2 #16: Looking for a beta tester

roleoroleo, 2020-10-29:
I'm try to refactor the code (RTSP and snapshot) with the aim of having a single piece of code.
I need a beta tester with a y21ga and another with a r30gb.
I don't want to release a new bugged version...

[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5459438/rRTSPServer.gz)
[snapshot.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5459458/snapshot.tar.gz)

There are no differences with the released version, only code refactoring.

mcanteri, 2020-10-29:
> I'm try to refactor the code (RTSP and snapshot) with the aim of having a single piece of code.
> I need a beta tester with a y21ga and another with a r30gb.
> I don't want to release a new bugged [...]

I'm moving them to cam. Should be useful that gunzip or gzip -d would be there or work.
gzip isn't working:
gzip -d rRTSPServer.gz
gzip: applet not found
Will transfer them already uncompressed.

roleoroleo, 2020-10-29:
Remember to source the env:
`. /tmp/sd/yi-hack/script/env.sh`

mcanteri, 2020-10-29:
snapshot is working, both streams and on page.
RTSP server is working, high and low but not audio stream or audio in video.

roleoroleo, 2020-10-29:
So it works worse than released.

Does audio work withe the released version?

mcanteri, 2020-10-29:
My bad, I think that the integration with HA works as before.

I think it works the same more or less: maybe if you tell me what to pay attention more, I can give you maybe more specific feedback.  With the release 0.1.2 the audio doesn't work.

mcanteri, 2020-10-29:
Sorry, I remembered now that you said it was a refactor work. Ok, I think all still works as before.

roleoroleo, 2020-10-29:
Thank you.
Maybe @artemy ...

HuangRicky, 2020-10-30:
Hi @roleoroleo I have a **[Yi Outdoor 1080p](https://www.amazon.com/YI-Waterproof-Surveillance-Detection-Deterrent/dp/B01CW49AGG)**

Does this fw work on this cam? I can also provide it for testing if you tell me what I need to do: I assume I shall flash it and then expose the ssh thru port-forwarding? Thanks!
the serial number starts with 9CUS

roleoroleo, 2020-10-30:
Make a dump of the flash.

pep4n, 2020-10-30:
> Hi @roleoroleo I have a **[Yi Outdoor 1080p](https://www.amazon.com/YI-Waterproof-Surveillance-Detection-Deterrent/dp/B01CW49AGG)**
>
> Does this fw work on this cam? I can also provide it for testing [...]

this is what roleoroleo is referring to: 
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Dump-your-backup-firmware-(SD-card)

HuangRicky, 2020-10-31:
Hello @roleoroleo  and @pep4n . I wasn't able to use your script to dump FW. However, I was able to flash yi-hack-v4  from [here](https://github.com/TheCrypt0/yi-hack-v4/wiki/Getting-Started) and then I can SSH into it and dump the FW.
Now I have them zipped and attach here:
[backup_yi_outdoor.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5468227/backup_yi_outdoor.zip)

Note: this is after I flashed yi-hack-v4. Not sure if this can help. Thank you!

roleoroleo, 2020-10-31:
If your cam supports yi-hack-v4 you can't use this hack.
The hardware platform is different.

wimpie007, 2020-10-31:
@roleoroleo 
i have a dome guard (r30gb) and running 

Firmware Version | 0.1.2
Base Version | 9.0.05.32_202004101552
Model Suffix | r30gb

enabled swap.

only low-res snapshot and stream working.

roleoroleo, 2020-10-31:
Try the archive in the 1st post.

wimpie007, 2020-10-31:
snapshot works!
low AND high-res.

rtsp: low works fine.
high works for a few seconds (even refresh rate is ok), but after a while (5s?) it starts going wrong. garbled image, freezing.

roleoroleo, 2020-10-31:
Please try this for rtsp:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/10#issuecomment-717484211

wimpie007, 2020-10-31:
@roleoroleo  
rtsp_server from #10 seems to be working a bit longer on high-res before starting to garbling, but definately not ok.

mcanteri, 2020-10-31:
> @roleoroleo
> rtsp_server from #10 seems to be working a bit longer on high-res before starting to garbling, but definately not ok.

Great test, ok probably now we reached the level where Roleo have to make work the software for this model r30gb and the other y21ga
Hope the refactoring will make it easier :) :+1:

artemy, 2020-10-31:
> snapshot works!
> low AND high-res.
> 
> rtsp: low works fine.
> high works for a few seconds (even refresh rate is ok), but after a while (5s?) it starts going wrong. garbled image, freezing.

I have the same results, but only using files from this thread.

roleoroleo, 2020-10-31:
I need a dump of the high res stream for r30gb model.
Could you please run h264grabber and share with me the stream?
[h264grabber.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5469100/h264grabber.gz)

`h264grabber -m r30gb -r high -d > stream.h265`

I hope the cam is not in the bathroom :)

wimpie007, 2020-10-31:
@roleoroleo 
i grabbed a few minutes of high-res footage.
can play it perfectly on the laptop!
Still need it?

roleoroleo, 2020-10-31:
Try to process it with ffmpeg and check if there are errors.

An anomaly I found in the stream is that the vps comes before the sps and pps.
I don't know if this can be a problem.

wimpie007, 2020-10-31:
no idea how to check it. can upload it to wetransfer. can i send you the link?

roleoroleo, 2020-10-31:
Ok, thanks.

wimpie007, 2020-10-31:
@roleoroleo 
wetransfer link:
https://wetransfer.com/downloads/2a76363e51792ff36af40550b166135420201031184154/24b00df27ef21139a58b0ae0f9c6433f20201031184254/e4eb33

roleoroleo, 2020-10-31:
Copied, thanks.

roleoroleo, 2020-10-31:
VLC and ffmpeg read it correctly.

wimpie007, 2020-10-31:
mpc played it perfect also...

wimpie007, 2020-11-01:
buffer size to big when streaming?
There is a big delay between real-camera and 'rtsp stream'.
That huge delay is not visible in the yi-app.

roleoroleo, 2020-11-01:
Yi app is a completely different technology.
It's not streamed with rtsp.
I don't know if the problem is live555 library.
But I can't find a better server.

roleoroleo, 2020-11-03:
Another try please.
I fixed some lines of code.
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5481202/rRTSPServer.gz)

mcanteri, 2020-11-03:
> Another try please.
> I fixed some lines of code.
> [rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5481202/rRTSPServer.gz)

Both test or only r30gb ?

mcanteri, 2020-11-03:
High and low streams looks clean and with audio 👍 
Audio stream works 👍 
Integration with HA is fluent, it miss audio don't know why and testing now the production of motion file by HA, if there are some not playable.

roleoroleo, 2020-11-03:
Both models, if possible.

wimpie007, 2020-11-03:
test on r30gb:
*audio seems to work!* (not in HA)
for the rest: low-stream works like before, high-stream is worse. Almost impossible to start playing the high-res stream. (it worked 1 time for some frames only)

roleoroleo, 2020-11-03:
So...
Y21ga: ok.
r30gb: problems with high res.

I will try to fix the problem.

mcanteri, 2020-11-03:
There are some artefacts in high resolution on the y model too from time to time.

roleoroleo, 2020-11-04:
> There are some artefacts in high resolution on the y model too from time to time.
> I can see now that in low light there are much more artefacts than before.

I thought I had improved, not got worse :)

I have another version where I moved VPS before SPS (now the sequence is wrong).
I need a test (only r30gb), thank you.
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5490274/rRTSPServer.gz)

wimpie007, 2020-11-05:
low-stream ok as before, high-stream bad, got it to stream high-res **once** for 300 frames after a lot of trying, then freezes up.

roleoroleo, 2020-11-05:
One more time around the block.

roleoroleo, 2020-11-05:
@wimpie007 
Could you please run this version with debug option and share the output?
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5496292/rRTSPServer.gz)

```
RRTSP_DEBUG=1 RRTSP_MODEL=r30gb RRTSP_RES=both RRTSP_AUDIO=yes /tmp/sd/yi-hack/bin/rRTSPServer
```

wimpie007, 2020-11-05:
@roleoroleo 
low-stream working as before, (altough no audio, don't know if it worked in the previous version also, didnt pay attention to it)
high-res is better! sometimes, i can stream 1 minute of high-res. after that, slows down (1 frame/s) en then stops completely.
Logs doesnt change much when things go bad...

`34:` frame_len: 3247 - cb_current->size: 262144
1604603659234: frame_len: 3247 - frame_counter: 38816 - resolution: 1080
1604603659259: frame_len: 855 - cb_current->size: 49152
1604603659259: frame_len: 855 - frame_counter: 38185 - resolution: 360
1 - fFrameSize 855 - fMaxSize 66878 - fLimitNumBytesToStream 0
1 - Completed
0 - fFrameSize 855 - fMaxSize 66023 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
1604603659284: frame_len: 3174 - cb_current->size: 262144
1604603659284: frame_len: 3174 - frame_counter: 38817 - resolution: 1080
1604603659310: frame_len: 856 - cb_current->size: 49152
1604603659310: frame_len: 856 - frame_counter: 38186 - resolution: 360
1 - fFrameSize 856 - fMaxSize 66023 - fLimitNumBytesToStream 0
1 - Completed
0 - fFrameSize 856 - fMaxSize 65167 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
1604603659335: frame_len: 3098 - cb_current->size: 262144
1604603659335: frame_len: 3098 - frame_counter: 38818 - resolution: 1080
1604603659335: frame_len: 817 - cb_current->size: 49152
1604603659335: frame_len: 817 - frame_counter: 38187 - resolution: 360
1 - fFrameSize 817 - fMaxSize 65167 - fLimitNumBytesToStream 0
1 - Completed
0 - fFrameSiz``

roleoroleo, 2020-11-05:
If it's possible I need a log > 2 seconds.
Every 2 seconds there is a SPS frame (log similar to "h265 SPS detected, writing VPS").

Are there any lines like this?
"frame size exceeds buffer size"

wimpie007, 2020-11-06:
"frame size exceeds buffer size"

No, not in the log. Here is a log of a few minutes.
https://wetransfer.com/downloads/bc380d7c0eafd460cca022906ea2769120201106073120/fbb8e1c633223d0b459856461f8193c320201106073153/6de74a

roleoroleo, 2020-11-06:
@wimpie007 
Looking at your log there are a lot of sync problem.
The frame counter jumps back incorrectly, probably when the buffer is circulating.
I removed VPS reordering that I previously added, maybe this is the problem.
Try again with debug option please.
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5502695/rRTSPServer.gz)

wimpie007, 2020-11-07:
unfortunately, no big difference.
maybe (subjective?) a little easyer to start it streaming, but after 1.30 min. or something, it stops...
log:
https://wetransfer.com/downloads/5f8e85fa40ff6b2867363fef808f6f9f20201107084535/509a1f0039e1992a7f9e1364f8aa803420201107084616/c43170

Thanks for the great work @roleoroleo

roleoroleo, 2020-11-07:
I will add a frame counter check.

roleoroleo, 2020-11-07:
Added a check on the frame counter:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5505300/rRTSPServer.gz)

wimpie007, 2020-11-07:
mmm, at first try, with VLC, it looked promising. I could stream high-res, after 90 seconds, it would stop, but pressing STOP  and PLAY again would start te stream again.
However... the process stops often. Debug log just stops, not killing the process (so watchdog doesnt see anything wrong)
Low-res suffers the same problem now also.
so: high-res: better if process doesnt stop. Low-res: worse then before. (it was working ok)

log of session 'stops' logging/streaming...

`1604780780322: frame_len: 949 - cb_current->size: 49152
1604780780322: frame_len: 949 - frame_counter: 24814 - resolution: 360
1604780780322: frame_len: 3046 - cb_current->size: 262144
1604780780322: frame_len: 3046 - frame_counter: 24979 - resolution: 1080
1604780780322: frame_len: 19 - cb_current->size: 49152
1604780780322: frame_len: 19 - frame_counter: 24815 - resolution: 360
1604780780398: frame_len: 2468 - cb_current->size: 262144
1604780780398: frame_len: 2468 - frame_counter: 24980 - resolution: 1080
1604780780424: frame_len: 878 - cb_current->size: 49152
1604780780424: frame_len: 878 - frame_counter: 24818 - resolution: 360
0 - fFrameSize 74390 - fMaxSize 75609 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
1 - fFrameSize 19862 - fMaxSize 75609 - fLimitNumBytesToStream 0
1 - Completed
0 - fFrameSize 19862 - fMaxSize 55747 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
1604780782335: frame_len: 2394 - cb_current->size: 262144
1604780782335: frame_len: 2394 - frame_counter: 24981 - resolution: 1080
1604780782335: frame_len: 878 - cb_current->size: 49152
1604780782335: frame_len: 878 - frame_counter: 24819 - resolution: 360
1 - fFrameSize 2394 - fMaxSize 55747 - fLimitNumBytesToStream 0
1 - Completed
0 - fFrameSize 2394 - fMaxSize 53353 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
1604780783315: frame_len: 2185 - cb_current->size: 262144
1604780783315: frame_len: 2185 - frame_counter: 24982 - resolution: 1080
1 - fFrameSize 2185 - fMaxSize 53353 - fLimitNumBytesToStream 0
1 - Completed
0 - fFrameSize 2185 - fMaxSize 51168 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
1604780783416: frame_len: 870 - cb_current->size: 49152
1604780783416: frame_len: 870 - frame_counter: 24820 - resolution: 360
`

roleoroleo, 2020-11-08:
@wimpie007 
I fixed a few lines of code:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5505963/rRTSPServer.gz)

wimpie007, 2020-11-08:
nope, not better...
same result more or less. (high and low stream very problematic)
i even got one exception thrown.

every few seconds i see this: (fram_len very low?)

1604828714633: frame_len: 4209 - frame_counter: 23047 - resolution: 1080
1604828714714: frame_len: 4105 - cb_current->size: 262144
1604828714715: frame_len: 4105 - frame_counter: 23048 - resolution: 1080
1604828714715: frame_len: 56 - cb_current->size: 262144
1604828714715: frame_len: 56 - frame_counter: 23049 - resolution: 1080
1604828714715: frame_len: 13 - cb_current->size: 262144
1604828714715: frame_len: 13 - frame_counter: 23050 - resolution: 1080
1604828714716: frame_len: 28 - cb_current->size: 262144
1604828714716: frame_len: 28 - frame_counter: 23051 - resolution: 1080
1604828714743: frame_len: 142753 - cb_current->size: 262144
1604828714743: frame_len: 142753 - frame_counter: 23052 - resolution: 1080
1604828714769: frame_len: 2979 - cb_current->size: 262144
1604828714769: frame_len: 2979 - frame_counter: 23053 - resolution: 1080
1604828714844: frame_len: 3806 - cb_current->size: 262144
1604828714845: frame_len: 3806 - frame_counter: 23054 - resolution: 1080
1604828714895: frame_len: 3991 - cb_current->size: 262144

roleoroleo, 2020-11-08:
The log is correct.
Every 2 seconds you should read this (GOV length is 40 @20 fps).
56 is the SPS
13 is the PPS
28 is the VPS
142K is the I-frame

Could you send me a log of a longer sequence?

wimpie007, 2020-11-08:
https://wetransfer.com/downloads/1435d136cfcddc3f70046eba6299a2fe20201108102143/9f0a0399d36f28dc0c719c45d592802220201108102157/7c24e5
in this session, i could not success in streaming the high-res stream... not even once.

roleoroleo, 2020-11-08:
Other fixes:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5506213/rRTSPServer.gz)

wimpie007, 2020-11-08:
nope, not better unfortunately. except that i had AUDIO only stream at one moment. (no image, only audio)

roleoroleo, 2020-11-08:
Damn...
Please send me a log.

wimpie007, 2020-11-08:
can i send it directly - mail?

roleoroleo, 2020-11-08:
Yes.
Use the email linked with the paypal account.

hamburgdude, 2020-11-11:
Hey, excellent work. RTSP stream works great, no issues.

Just a remark for PTZ: Got the message "PTZ functions are not available for this camera model" after short popup of the move pad. 

Model y21ga, Firmware 0.1.2, Base 9.0.19.02_202007141512

jjleto, 2020-11-20:
Yi Dome Guard fw 9.0.05.32_202004101552 s/n KFUSY with r30gb_0.1.2
What works:
- High res stream
- Low res stream
- Low res snapshot
- motion events recording

 What’s not working:
- High res snapshot (green screen)
- WiFi Strength (0%)
- PTZ (perhaps) reversed movement of cam respect to the arrows direction

[4 more replies]

closed 2020-12-11 by roleoroleo