roleoroleo/yi-hack-Allwinner-v2 #17: How to disable "Motion event recordings"?

marcomehnert, 2020-10-30:
Isn't it possible to fully disable the recording of motion events? Is the configuration done within the web interface not fully implemented yet? The camera works well and I could connect two of the with my motioneye on Raspberry Pi. But I dont want to record motion events on the SD card of the cameras.

Thanks in advance and also for the great work on that hack!

Best

Marco

roleoroleo, 2020-10-30:
The hack doesn't allow to switch off the recording.
But probably you can do it with the app, I don't remember exactly.

If using the app is not possible, you could try to kill mp4record.
Create a shell script
/tmp/sd/yi-hack/startup.sh
with this content:
```
#!/bin/sh

killall mp4record
```

closed 2020-12-27 by roleoroleo

brothert, 2021-01-30:
mp4record will be restarted within a few seconds by watch_process unless you kill that too.

mp4record can be blocked by creating a file:

mv /tmp/sd/record /tmp/sd/record.bak
touch /tmp/sd/record

Jon8RFC, 2021-12-05:
> mp4record will be restarted within a few seconds by watch_process unless you kill that too.
> 
> mp4record can be blocked by creating a file:
> 
> mv /tmp/sd/record /tmp/sd/record.bak touch /tmp/sd/record

Does that actually disable reads/writes to the sd card, or just keep it looping and reading/writing to the card and failing, instead?
My instinct is that it just fails over and over again, but it doesn't _appear_ to be using the sd card.

brothert, 2021-12-05:
I don't think so. mp4record obviously checks whether /tmp/sd/record exists and tries to write a file in it, which immediately fails in open() because record itself is a file, or it realizes that record ist not a directory and tries to mkdir() it which immediately fails as well. In both cases the SD card is not written to.

atlaware, 2023-07-26:
Related to this "feature request", there is some news with all changes comes in the last release? Can be disabled motion recording without disable the camera at all?
I use this hack with home-assistant and I need, for a "privacy compromise" to disable recording, but I want that live video continue to work for using it ad "baby monitor", when baby sleep or stay alone in some parts of house.

Thank you.

roleoroleo, 2023-07-26:
> Related to this "feature request", there is some news with all changes comes in the last release? Can be disabled motion recording without disable the camera at all?

Try this.
Create a shell script
/tmp/sd/yi-hack/startup.sh
with this content:

#!/bin/sh

killall watch_process
killall mp4record

> I use this hack with home-assistant and I need, for a "privacy compromise" to disable recording, but I want that live video continue to work for using it ad "baby monitor", when baby sleep or stay alone in some parts of house.

If you use the "Switch on/off the camera" switch, this should disable local recording. I don't remember exactly but it should.