roleoroleo/yi-hack-Allwinner-v2 #18: hack not working on dome guard

accelle17, 2020-11-01:
Details:

Firmware Version | 0.1.2
Base Version | 9.0.05.32_202004101552
Model Suffix | r30gb
Model KFUSY36

I already copied folders yi-hack, newbackup, and Factory as well as lower_half_init.sh but it doesn't work. Before this, I am able to backup using the procedure provided.

On the logs located on the root, there's log_first_login.tar, log_login.tar.gz, and log_wifi_connected.tar.gz. I can see httpd -p 8080 on the ps1.txt,, telnetd and dropbear.
Whats logs can I check further to see whats wrong?

wimpie007, 2020-11-01:
That's strang,

Firmware Version | 0.1.2
Base Version | 9.0.05.32_202004101552
Model Suffix | r30gb
Serial Number | HFUSY36

Working fine here.
Can't you browse to : ip:8080?

accelle17, 2020-11-01:
> That's strange,
> 
> Firmware Version | 0.1.2
> Base Version | 9.0.05.32_202004101552
> Model Suffix | r30gb
> Serial Number | HFUSY36
> 
> Working fine here.
> Can't you browse to : ip:8080?

No http:8080, ssh and telnet doesnt work. I could ping the ip.

wimpie007, 2020-11-01:
can you see the recordings in the sd-card?
Before copying the hack (3 directories and lower_half_init.sh to root), did you delete all dirs/files?

accelle17, 2020-11-01:
> can you see the recordings in the sd-card?
> Before copying the hack (3 directories and lower_half_init.sh to root), did you delete all dirs/files?

Yes, can see the recordings. I tried with fresh sd card and also tried deleting and replacing the 3 directories  after it was inserted.

adrianperezferro, 2020-11-05:
Just installed on my Home Guard and working ok. It failed when I had the Yi Home app opened on my phone. As soon as I closed it, it just started working.

Now i'm seeing that RTSP seems to freeze and the HQ snapshot has a codec problem as it is all green.

Good job @roleoroleo !!!

roleoroleo, 2020-11-05:
Try the beta version in the other issue.

adrianperezferro, 2020-11-05:
Awesome! Now it works well. Thank you so much!.

accelle17, 2020-11-08:
Removing the internet on the device made the hack worked.

closed 2020-11-08 by accelle17