roleoroleo/yi-hack-Allwinner-v2 #19: Remove YI watermark from frames

c3kkos, 2020-11-01:
As topic says, is it possible to remove the YI logo that appears in the lower left corner of every frames?

roleoroleo, 2020-11-02:
I will check it but if it's like in the other cam there is nothing to do.
Read this issue: https://github.com/roleoroleo/yi-hack-Allwinner/issues/34

c3kkos, 2020-11-02:
Ok i read.

Si even if it is not possible to disable the watermark maybe there's a trick to make it disappear.

I see from frames that the white color is somehow blended with the pic content. Like the blend mode we have in photoshop/gimp: if blend mode is "screen" or "add" maybe we can just fill the bmp full black.

Well this reminds me the UFO wall poster "i want to believe"

c3kkos, 2020-11-02:
So here i attach the modified bmp, which is filled by the gray color of the original.

It should blend with frame content but.... i'm unable to test due to main.bmp is on a read only filesystem.

There's a trick to write onto it?

roleoroleo, 2020-11-02:
> There's a trick to write onto it?

If you dont' want to overwrute the whole partition, the better solution is to use mount with bind option.
Check out this file: https://github.com/roleoroleo/sonoff-hack/blob/master/sdhack/boot.sh

c3kkos, 2020-11-03:
Sorry to bother you Roleo, 

binding the file is ok.. even without upload the modified file, i ran (for test):

```
root@yicam1:/tmp# . /tmp/sd/yi-hack/script/env.sh
root@yicam1:/tmp# touch /tmp/main.bmp
root@yicam1:/tmp# mount --bind /tmp/main.bmp /home/app/main.bmp
```

and logically the logo is still there cuz it is loaded into ram. 

After binding the new main.bmp we need to restart the specific service.

roleoroleo, 2020-11-03:
I think you can't restart processes, the system probably crashes.
You have to add the bind to the startup scripts before rmm starts.
For example in /tmp/sd/yi-hack/script/system.sh

c3kkos, 2020-11-03:
ok i'll try asap

roleoroleo, 2020-11-03:
Could you take a snapshot?

c3kkos, 2020-11-03:
sure, here we have the grey watermark.

now i'll try if black is better.
[image]

c3kkos, 2020-11-03:
This is the results with pure black bmp.. not so different. 
[image]

and this is the black bmp

[black_main.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5483155/black_main.zip)

roleoroleo, 2020-11-03:
> And.. strange enough, the BMP after use by the cam, becomes unreadable..
> 
> if i try to force open and save it turns black. (irfanview and gimp)

Maybe it's busy.

c3kkos, 2020-11-04:
> > And.. strange enough, the BMP after use by the cam, becomes unreadable..
> > if i try to force open and save it turns black. (irfanview and gimp)
> 
> Maybe it's busy.

But windows preview (thumbnail) works! Gimp reject it and ifranview opens it correctly! But when i overwrite the bmp within irfanview, the new picture is pure black!

I double checked this since its a weird behaviour.

roleoroleo, 2020-11-04:
> Roleo thank you again. I'm using your stuff once again though, i remember you for your agpf firmware. What a good time!

It's been a long time...

EsEnZeT, 2020-12-23:
@c3kkos @roleoroleo I've installed the hack FW on my camera but switching watermark ON/OFF doesn't change anything (rtsp / /cgi-bin/snapshot.sh). Do I understand correctly I can only replace (bind mount) some file in this case?

[image]

roleoroleo, 2020-12-23:
Where do you switch on/off watermark?

EsEnZeT, 2020-12-23:
Using WEB GUI:
- http://{CAM_IP}:8080/index.html?page=configurations (ONVIF watermark)

Also tried to force no watermark by visiting:
- http://192.168.1.8:8080/cgi-bin/snapshot.sh

roleoroleo, 2020-12-23:
Yes, it's date and time in the snapshot.

closed 2020-12-27 by roleoroleo

chross, 2024-02-13:
Just a little addition:
there is also a sub.bmp with 32x32 dimensions, that needs to be rebound using the same method.