roleoroleo/yi-hack-Allwinner-v2 #21: Boot stopped after reboot  by "Unbrick the cam"

romakrau, 2020-11-03:
Hello, I've used the backup discribed at "unbrick the cam" and reset the cam. After the first time it's works fine. At the second time the boot process stopped and the cam is blocked. Can anybody help?

[putty1.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5480954/putty1.log)

Greetings Roman

Sorry for my GermanEnglish.

romakrau, 2020-11-03:
Additional log file.

[putty.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5482730/putty.log)

roleoroleo, 2020-11-03:
"Unbrick the cam" doesn't backup your system.
It restores your firmware if you have a backup made previuosly.

romakrau, 2020-11-04:
The camera is still alive. It take a lot of patience. I guess the mistake I did is that I leave the SD-Card in. The Camera tried to mount the card and was confused. Me too ;-) 
How can I use the hack y21ga... in my camera. Have I left the sd-card in?
Greetings Roman

c3kkos, 2020-11-04:
I suggest to improve your english! Hehe.. well yes my friend, you need to have a SD card IN, all the time.

Since this hack do not replace any static firmware, instead, the new one is loaded on boot time, from the SD card.

romakrau, 2020-11-04:
Sorry I didn't improve my English but I improve my knowledge how the software works. Now it works fine for the moment.

closed 2020-11-04 by romakrau