roleoroleo/yi-hack-Allwinner-v2 #22: missing ftpput

iupsilon, 2020-11-04:
Hi,
I succesfully installed y21ga on Yi 1080p Home BFUS (9.0.19*)
After a bit of struggling I figured out why ftp upload doesn't work.

The script ftppush.sh calls ftpput but there is no such executable in export PATH directory list.
Il looks like the file is missing.

I will give you feedback as soon as you provide a new version of the firmware.
Thanks a lot.

roleoroleo, 2020-11-04:
system.sh runs ftppush.sh with the full path, it doesn't need the PATH variable.
Currently the script doesn't work because 2 busybox applets are missing.
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/5fe19b4113e57e8f07440f8034efa11fb184244d

adrianofoschi, 2020-11-13:
The same issue there.
I recompiled the latest sources from master but the ftp upload does not work...
How can I discover the error?

roleoroleo, 2020-11-13:
Try to run it manually:
```
# . /tmp/sd/yi-hack/script/env.sh
# /tmp/sd/yi-hack/script/env.sh start
```

adrianofoschi, 2020-11-13:
The commands output:

```
root@cam-3:~# . /tmp/sd/yi-hack/script/env.sh
root@cam-3:~# /tmp/sd/yi-hack/script/ftppush.sh start
2020-11-13 [22-33-07] [INFO] === SERVICE START ===
2020-11-13 [22-33-07] [INFO] checkFiles
/tmp/sd/yi-hack/script/ftppush.sh: line 238: nc: not found
/tmp/sd/yi-hack/script/ftppush.sh: line 238: ftpput: not found
[ERROR] uploadToFtp: ftpput FAILED.
2020-11-13 [22-33-07] [ERROR] checkFiles: uploadToFtp FAILED - [/tmp/sd/record/2020Y11M13D20H/54M00S60.mp4].
/tmp/sd/yi-hack/script/ftppush.sh: line 238: nc: not found
/tmp/sd/yi-hack/script/ftppush.sh: line 238: ftpput: not found
[ERROR] uploadToFtp: ftpput FAILED.
2020-11-13 [22-33-08] [ERROR] checkFiles: uploadToFtp FAILED - [/tmp/sd/record/2020Y11M13D20H/55M00S60.mp4].
/tmp/sd/yi-hack/script/ftppush.sh: line 238: nc: not found
/tmp/sd/yi-hack/script/ftppush.sh: line 238: ftpput: not found
[ERROR] uploadToFtp: ftpput FAILED.
2020-11-13 [22-33-08] [ERROR] checkFiles: uploadToFtp FAILED - [/tmp/sd/record/2020Y11M13D20H/56M00S60.mp4].
```

roleoroleo, 2020-11-13:
nc and ftpput?!?
Are they in the /tmp/sd/y-hack/usr/bin folder?

adrianofoschi, 2020-11-13:
No they didn't...but now after a "git pull" I see I've pulled a modified version of `install.busybox`.
Now I'm trying to recompile the latest sources. Thanks you roleo

adrianofoschi, 2020-11-14:
OK now It works. You can close the issue thanks you :)

closed 2020-11-14 by roleoroleo

szczyglik, 2020-11-27:
Hi Guys. May I ask You to update latest compiled sources? Unfortunately do not have access to Linux except raspberry. Compiling on ubuntu pi 32bit isn't working.

szczyglik, 2020-11-27:
> OK now It works. You can close the issue thanks you :)

Hi. May I ask You to send me latest compiled sources? Unfortunately I do not have access to Linux except raspberry. Compiling on ubuntu pi 32bit isn't working.