roleoroleo/yi-hack-Allwinner-v2 #23: Switch on the Camera settings per hhtp?

romakrau, 2020-11-07:
Hi, I want use the camera-hack to control the camera with my smart software FHEM. So I would like ask if there is a way to switch the camera settings "Switch on/off the camera" and "Save video when a motion is detected" per web service like the system settings (http://IP:8080/cgi-bin/get_configs.sh?conf=CONF).
Greetings Roman

roleoroleo, 2020-11-07:
I didn't test it because I don't have this cam.
But you could try:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Web-services-description#camera-settings

romakrau, 2020-11-08:
This is a great function, but is there a way to get only a sign when a motion is detected.
Greetings Roman

roleoroleo, 2020-11-08:
Mqtt

romakrau, 2020-11-09:
Yes I'had seen it. MQTT is working. But I'm looking for a sign without video recording.  Thanks a lot for your answers. Greeitngs Roman

roleoroleo, 2020-11-09:
What do you mean with "sign".
The cam sends a message when the motion starts, even it it's not recording.

romakrau, 2020-11-10:
Sorry I can commit this behaivor. My Cam recorded allways if a motion is deteced. I can't find a switch for recording as described in the camera settings menu. Where is the switch to enable the recording?

If enabled, video will be saved only when a motion is detected.    
If disabled, video will be always saved.
(It takes effect only if recording is enabled)

roleoroleo, 2020-11-10:
The cam doesn't record when "Disable Cloud" is on and "Recording without Cloud" is off.

romakrau, 2020-11-10:
Thank you for your help. I didn't understand "Disable Cloud" miss.

closed 2020-11-10 by romakrau