roleoroleo/yi-hack-Allwinner-v2 #25: Unable to compile www

adrianofoschi, 2020-11-12:
Hey, thanks for you work.
Today I'm trying to compile the firmware becouse I need the ftp upload feature that doesn't work in the first release.

I'm following this guide: https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Build-your-own-firmware

As compile environment I'm using a debian in a docker container. 
Some points of attention:

1. I'm using "stretch" instead of "buster" becouse with "buster" it is missing libtidy5
2. Probably you need to add "wget" package in needed packages list becouse it is required by "alsa"

The problem the compile process stops at "www" compilation.
This is the output log:

```
MOD_DIR:        /yi-hack-Allwinner-v2/src/www
MOD_NAME:       www
MOD_INIT:       init.www
MOD_COMPILE:    compile.www
MOD_INSTALL:    install.www

Initializing www...

Compiling www...

Compressing CSS files...
Compressing JS files...
Exception in thread "main" java.lang.reflect.InvocationTargetException
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:498)
        at com.yahoo.platform.yui.compressor.Bootstrap.main(Bootstrap.java:21)
Caused by: java.util.MissingResourceException: Can't find bundle for base name org.mozilla.javascript.resources.Messages, locale en
        at java.util.ResourceBundle.throwMissingResourceException(ResourceBundle.java:1581)
        at java.util.ResourceBundle.getBundleImpl(ResourceBundle.java:1396)
        at java.util.ResourceBundle.getBundle(ResourceBundle.java:854)
        at org.mozilla.javascript.ScriptRuntime$DefaultMessageProvider.getMessage(ScriptRuntime.java:3608)
        at org.mozilla.javascript.ScriptRuntime.getMessage(ScriptRuntime.java:3592)
        at org.mozilla.javascript.ScriptRuntime.getMessage0(ScriptRuntime.java:3540)
        at org.mozilla.javascript.Parser.addError(Parser.java:145)
        at org.mozilla.javascript.Parser.reportError(Parser.java:160)
        at org.mozilla.javascript.Parser.statementHelper(Parser.java:1285)
        at org.mozilla.javascript.Parser.statement(Parser.java:726)
        at org.mozilla.javascript.Parser.parseFunctionBody(Parser.java:482)
        at org.mozilla.javascript.Parser.function(Parser.java:611)
        at org.mozilla.javascript.Parser.parseFunctionBody(Parser.java:479)
        at org.mozilla.javascript.Parser.function(Parser.java:611)
        at org.mozilla.javascript.Parser.primaryExpr(Parser.java:2255)
        at org.mozilla.javascript.Parser.memberExpr(Parser.java:1974)
        at org.mozilla.javascript.Parser.unaryExpr(Parser.java:1832)
        at org.mozilla.javascript.Parser.mulExpr(Parser.java:1761)
        at org.mozilla.javascript.Parser.addExpr(Parser.java:1742)
        at org.mozilla.javascript.Parser.shiftExpr(Parser.java:1722)
        at org.mozilla.javascript.Parser.relExpr(Parser.java:1696)
        at org.mozilla.javascript.Parser.eqExpr(Parser.java:1652)
        at org.mozilla.javascript.Parser.bitAndExpr(Parser.java:1641)
        at org.mozilla.javascript.Parser.bitXorExpr(Parser.java:1630)
        at org.mozilla.javascript.Parser.bitOrExpr(Parser.java:1619)
        at org.mozilla.javascript.Parser.andExpr(Parser.java:1607)
        at org.mozilla.javascript.Parser.orExpr(Parser.java:1595)
        at org.mozilla.javascript.Parser.condExpr(Parser.java:1578)
        at org.mozilla.javascript.Parser.assignExpr(Parser.java:1563)
        at org.mozilla.javascript.Parser.expr(Parser.java:1542)
        at org.mozilla.javascript.Parser.primaryExpr(Parser.java:2412)
        at org.mozilla.javascript.Parser.memberExpr(Parser.java:1974)
        at org.mozilla.javascript.Parser.unaryExpr(Parser.java:1832)
        at org.mozilla.javascript.Parser.mulExpr(Parser.java:1761)
        at org.mozilla.javascript.Parser.addExpr(Parser.java:1742)
        at org.mozilla.javascript.Parser.shiftExpr(Parser.java:1722)
        at org.mozilla.javascript.Parser.relExpr(Parser.java:1696)
        at org.mozilla.javascript.Parser.eqExpr(Parser.java:1652)
        at org.mozilla.javascript.Parser.bitAndExpr(Parser.java:1641)
        at org.mozilla.javascript.Parser.bitXorExpr(Parser.java:1630)
        at org.mozilla.javascript.Parser.bitOrExpr(Parser.java:1619)
        at org.mozilla.javascript.Parser.andExpr(Parser.java:1607)
        at org.mozilla.javascript.Parser.orExpr(Parser.java:1595)
        at org.mozilla.javascript.Parser.condExpr(Parser.java:1578)
        at org.mozilla.javascript.Parser.assignExpr(Parser.java:1563)
        at org.mozilla.javascript.Parser.assignExpr(Parser.java:1569)
        at org.mozilla.javascript.Parser.expr(Parser.java:1542)
        at org.mozilla.javascript.Parser.statementHelper(Parser.java:1221)
        at org.mozilla.javascript.Parser.statement(Parser.java:726)
        at org.mozilla.javascript.Parser.parse(Parser.java:420)
        at org.mozilla.javascript.Parser.parse(Parser.java:378)
        at com.yahoo.platform.yui.compressor.JavaScriptCompressor.parse(JavaScriptCompressor.java:312)
        at com.yahoo.platform.yui.compressor.JavaScriptCompressor.<init>(JavaScriptCompressor.java:533)
        at com.yahoo.platform.yui.compressor.YUICompressor.main(YUICompressor.java:131)
        ... 5 more
```

roleoroleo, 2020-11-13:
If I remember correctly, I manually installed the new version contained in buster (2.4.8-2).
https://packages.debian.org/stable/yui-compressor

adrianofoschi, 2020-11-13:
OK BUR are you using buster? What about libtidy?

roleoroleo, 2020-11-13:
But probably I will move to closure-compiler.

roleoroleo, 2020-11-13:
I'm using stretch and libtidy is not installed.

adrianofoschi, 2020-11-13:
OK I succesfully compiled it using debian:stable docker image and libtidy5deb1.

Debian container commands:
```
docker pull debian:stable
docker run --name debian-stable -h stable -e LANG=C.UTF-8 -it debian:stable /bin/bash -l
```

Container inside commands
```
apt-get update
apt-get install git unzip build-essential bison++ bisonc++ libbison-dev autoconf autotools-dev automake libssl-dev zlibc zlib1g-dev libzzip-dev flex libfl-dev yui-compressor closure-compiler optipng jpegoptim libtidy5deb1 node-less sassc sass-spec libhtml-tidy-perl libxml2-utils rsync wget
git clone https://github.com/lindenis-org/lindenis-v536-prebuilt
mkdir -p /opt/yi/toolchain-sunxi-musl
cp -r lindenis-v536-prebuilt/gcc/linux-x86/arm/toolchain-sunxi-musl/toolchain /opt/yi/toolchain-sunxi-musl/
git clone https://github.com/roleoroleo/yi-hack-Allwinner-v2
cd yi-hack-Allwinner-v2/
git submodule update --init
./scripts/compile.sh
./scripts/pack_fw.all.sh
```

closed 2020-11-13 by adrianofoschi