roleoroleo/yi-hack-Allwinner-v2 #26: No discoverable / no video link by NVR

srcastor, 2020-11-15:
I just flash 0.1.2 to new BFUS cam (never used) with 9.0.19 firmware, all seems to work fine (http config page, settings...) however, is not founded by the NVR which has right now 4 onvif cameras configured and working.

If I add it manualy by IP, NVR seems read onvif information for model, number serial and that, but not video link is available, rtsp is working fine from VLC.

roleoroleo, 2020-11-15:
Is ws discovery enabled?

srcastor, 2020-11-15:
Yes, I forget to mention it, anyway, I tried any kind of combinations with no luck, high rtsp with high onvif, rtsp off and onvif on, onvif off and rstp on, both, low, disocvery off

[image]

roleoroleo, 2020-11-15:
WS Discovery is useful to discover onvif cam in your network.
If you disable it, your nvr can't get the cam automatically, you have to enter the IP manually.

Video is another problem. What protocol is able to decode your nvr?
This cam works with rtsp streams coded with h264 encoder.

Try to read the stream with vlc (both low and high res).
If it works, try to add the cam as a rtsp cam.

srcastor, 2020-11-15:
I tried with "ONVIF Device Manager" for windows, and all cams are detected, yi and my 4 onvif cams, I can play all the streams , I wonder what it's the difference, maybe onvif version? Unfortunately, my NVR can't add rstp cam directly, only onvif and supports h265 and h264 streams

closed 2022-01-20 by roleoroleo