roleoroleo/yi-hack-Allwinner-v2 #28: not installing on two BFUS

loginsistemas, 2020-11-17:
Hi.

I just received two Yi home 1080. Both are "BFUS..." and came with factory firmware 9.0.19.02_202007141512

I formatted a 16GB SD inside the camera. Copied files from yi-hack-Allwinner-v2 0.1.2 "y21ga_0.1.2.tgz".

But unfortunately firmware isn't installed. The first time that camera boots from microsd, the orange led blinks once (and only once). Then it become fixed again and quickly turns blinking blue and then fixed blue. The firm is not altered at all and the boot process doesn't take much more time than original.

Tested in both cameras with same results. It's not my first "hacked" yi, so i think i'm doing it in the right way. 

Any clue about what's my mistake?

P.D. Not having success with v2, I tried v1. I know, BFUS is for firm 18..., but i had to try. Off course, it didn't work neither.

roleoroleo, 2020-11-17:
> I formatted a 16GB SD inside the camera. Copied files from yi-hack-Allwinner-v2 0.1.2 "y21ga_0.1.2.tgz".

Did you decompress the archive?

loginsistemas, 2020-11-17:
> > I formatted a 16GB SD inside the camera. Copied files from yi-hack-Allwinner-v2 0.1.2 "y21ga_0.1.2.tgz".
> 
> Did you decompress the archive?

Oh, sure! I forgot to talk about this step.

d-paw, 2020-11-18:
So there are two of us, @loginsistemas 
I've got exactly the same problem as you do.
The orange led blinks once. Then it become fixed again and quickly turns blinking blue and actually connects to wifi. There's no "flashing" going on.

mcanteri, 2020-11-18:
Guys, did you try to connect to the camera http://192.168.X:X:8080/ ?  Put your address there

chechilll, 2020-11-18:
Cant flash my yi 1080p home camera BFUS.
I open the camera and got Y29GA_MB_V2.1 2020/04/07 with firmare 9.0.19.02_202007141512
Chipset QG2101A L1026AA
I tryed with diferents SD's, all in fat32 format.
Put the files downloaded from --> https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases --> file y21ga_0.1.2
I decompress the folder and put in the SD. Put the SD and power on the camera.
I've tried several times. Never work, always boot normal. Got ERR_CONNECTION_REFUSED if i put the http://IP_CAM:8080
SSh conection refused also

Please help, i'm desperated

[image]

d-paw, 2020-11-18:
> Guys, did you try to connect to the camera http://192.168.X:X:8080/ ? Put your address there

You are a legend, works like a charm!

roleoroleo, 2020-11-18:
@chechilll 
Factory.done means that the hack worked.
Check again your connection.

loginsistemas, 2020-11-18:
> Guys, did you try to connect to the camera http://192.168.X:X:8080/ ? Put your address there

OMG. Stupid mistake, I was triying port 80, like in other hacks. It's working fine. Maybe the confussion came from how easy is to apply the patch (just one blink doesn't looked to me like a firm update), and chinesse cloud continued to work as usual.
Now i have a CAM that doesn't connect to China or any other place.

Thank you and sorry for the waste of time. Thread can be closed.

closed 2020-11-18 by loginsistemas

chechilll, 2020-11-19:
Hello roleoroleo ,
It's not a connection problem. If the hack works, i've the followinf problems.
I'm reaching the IP address with ping.

[image]

I'm not able to connect with no one browser, as you can see it says connection refused as i said.

[image]

Also cant connect with SSH to port 22, connection refused

[image]

roleoroleo, 2020-11-19:
The correct address is http://192.168.1.244:8080

chechilll, 2020-11-19:
Thanks but i know. I read the posts :) I'm not newby
I tryed exactrly http://192.168.1.244:8080

[image]

roleoroleo, 2020-11-19:
Try to remove the hack and to apply it again.

chechilll, 2020-11-19:
Same result.
Can see the image via yi home app but with explorer connection refused, SSH connection refused