roleoroleo/yi-hack-Allwinner-v2 #29: FTP Upload creating DirTree is not working

inbux, 2020-11-17:
Hi,

creating a dir tree is not working in FTP Upload.

Looks like some commands are missing that are used in function lparentdir () of ftppush.sh.
xargs, dirname are not found.

In case there is CreatingDirTree = NO, there should be the dirname added to the filename, otherwise it is not possible to identify the date and hour of the mp4 file.
This means 15M46S14.mp4 should become  2020Y11M16D15H15M46S14.mp4

inbux, 2020-11-17:
I compiled to code by myself and added dirname to the installtion. Now it it working.

Simply add 

cp -f ./_install/yi-hack/bin/busybox ../../build/yi-hack/usr/bin/dirname

to the file 

/src/busybox/install.busybox

and compile.

---------------
If you want a long filename without creating directories, you can just modify this line in scripts/ftppush.sh:

else
		if ( ! ftpput -u "${FTP_USERNAME}" -p "${FTP_PASSWORD}" "${FTP_HOST}" "${FTP_DIR}**${FTP_DIR_HOUR}**$(lbasename "${UTF_FULLFN}")" "${UTF_FULLFN}" ); then
			logAdd "[ERROR] uploadToFtp: ftpput FAILED."
			return 1

(just insert the bold text)

---
Forget my remark about xargs  - this is because the PATH does not include the usr/bin in shell, but the ftppush.sh  adds the directories to the path.

roleoroleo, 2020-11-17:
Added dirname to busybox.

closed 2020-12-27 by roleoroleo