roleoroleo/yi-hack-Allwinner-v2 #30: Not able to update firmware

aalllop, 2020-11-19:
I have downloaded the files of the project but i'm not able to flash firmware. Are files correct? My cam is BFUS serial with firmware 9.0.19

roleoroleo, 2020-11-19:
Did you follow the update procedure described?

Murada99, 2020-11-19:
Exactly same serial BFUS and firmware 9.0.19 and same problem. The camera does not boot from the SD (formatted with the camera) but starts the YI solution.

aalllop, 2020-11-19:
The tgz files not contain a firmware like in the other projects. Maybe are they incorrect?

roleoroleo, 2020-11-19:
Yes, it's correct.

aalllop, 2020-11-19:
But have i got to compile the firmware and then flash it?

roleoroleo, 2020-11-19:
https://github.com/roleoroleo/yi-hack-Allwinner-v2#update-procedure

aalllop, 2020-11-19:
I have extracted y21ga_0.1.2.tgz on my sd card, pluged into the cam and powered on and nothings happended. The tar.gz file is full of folders and files, not 2 files with de firmware as the other git projects

roleoroleo, 2020-11-19:
This fw is different. It works using sd as a file system.
Because the home partition is read only.
A lot of files in the sd is normal and you can't remove the card.

Murada99, 2020-11-19:
@aalllop : same problem, I have extracted the same file to my SD Card (formatted with the camera) and nothing happens when I restart the camera.

jmtemmos, 2020-11-19:
same here. nothing happens. No update

aalllop, 2020-11-20:
I have tried again and the firmware was uploaded. The cam doesn't seem to upgrade but it did. To test it you can go to http://IP-CAM:8080/cgi-bin/snapshot.sh and you will se a snapshot

Murada99, 2020-11-20:
OK, it seems to work and I get access with http://ipcam:8080. 
Great !

gabaszo, 2020-11-20:
same here. nothing happens. No update  for Dome Guard model (Production model: YRS.3019)

Tomasinjo, 2020-11-20:
Can't get it to boot from the SD.
Model YRS.3019
R30GB_IN_V1_O 2019/11/04

Can someone check we have identical versions?

Murada99, 2020-11-21:
@Tomasinjo: When it says waiting to connect, follow the setup of the camera with the YI application. When it's done, insert the SD card and restart the camera. When restarted, check the the following address http://ip_of_camera:8080.

Tomasinjo, 2020-11-21:
Thanks, it works now. I just bought it and didn't even bother with their cloud service :)

closed 2020-12-27 by roleoroleo