roleoroleo/yi-hack-Allwinner-v2 #31: Yi Dome Guard compatability?

gabaszo, 2020-11-20:
Can I use this custom FW for Dome Guard model (Production model: YRS.3019)?

HoussamOtarid, 2020-11-23:
I would say yes, as I am doing that myself. The firmware is in beta version for the Dome Guard though.

Something additional you could check is that your product number (somewhere on the package label) should start with "*FUS" (mine starts with "KFUS"), and somewhere else on the package you should have the text "r30gb".

gabaszo, 2020-11-23:
I can find model: YRS.3019 and in SN r30gb
But when I copy all files from repot to root SD, and rebooting it-- nothing gonna happens.
SD is accessible for recording, and stock firmware is still used.

Murada99, 2020-11-23:
Didi you try to login with a browser at the following address http://cameraIP:8080 ?

gabaszo, 2020-11-24:
yes-- is not possible to open, means is not working

roleoroleo, 2020-11-25:
Try to chane the sd card.
r30gb hack should work.

gabaszo, 2020-11-25:
yep-- is working with a different SD. Thank you

tobal93, 2020-11-25:
> sí, está trabajando con una SD diferente. Gracias

What file should I download for dome guard?

roleoroleo, 2020-11-26:
If it's r30gb version, r30gb_0.1.2.tgz is your file.

tobal93, 2020-11-26:
> Si es la versión r30gb, r30gb_0.1.2.tgz es su archivo.

 it is a dome guard but it is KR30GB and when putting the file into it it does not start

roleoroleo, 2020-11-26:
Try the backup procedure and check if it works.

tobal93, 2020-11-26:
> Pruebe el procedimiento de copia de seguridad y compruebe si funciona.

I remove the card and if the camera works but with stock firmware, and when putting sd with hack it does not start

Technikhouse, 2021-03-23:
Hello,

I have buy (https://kamihome.com/dome-R30/)

can me someone explane to Hack this cam.

I Have checked with r30gb_0.15 but not working, the cam restart but the webserver is not working.

This files on the SD

[image]

Technikhouse, 2021-03-23:
[image]

roleoroleo, 2021-03-23:
If the folder Factory is not renamed to Factory,done the hack didn't work.
Probably it isn't an r30gb.

closed 2022-06-11 by roleoroleo