roleoroleo/yi-hack-Allwinner-v2 #33: yi dome guard

tecnologiaSsimple, 2020-11-21:
all functions are supported by yi dome GUARD KFUS except high resolution image capture. good job

roleoroleo, 2020-11-21:
Do you mean stream or snapshot?

tecnologiaSsimple, 2020-11-21:
Snaphot

tecnologiaSsimple, 2020-11-21:
Snaphot high resolution not working

roleoroleo, 2020-11-21:
Please try to run it manually.

ayasystems, 2020-11-26:
Same problem

High resolution is not working. I have done a backup prior to test the hack firmware i can send you if you want.

[image]
[image]
[image]

This is the ethernet chip of wired network
[image]

MDT File
[mtd.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5604284/mtd.txt)

Main chip
[image]

roleoroleo, 2020-11-26:
Try to run the command manually (with debug option) in ssh and post the output.

ayasystems, 2020-11-26:
What command? I'm very dummy on this things :D

roleoroleo, 2020-11-26:
imggrabber -r high -w > /tmp/test.jpg

ayasystems, 2020-11-27:
Hi

Where is located imggrabber??

```
root@yi-5cb1:/# imggrabber -r high -w > /tmp/test.jpg
-ash: imggrabber: not found
root@yi-5cb1:/# /home/yi-hack/bin/imggrabber -r high -w > /tmp/test.jpg
-ash: /home/yi-hack/bin/imggrabber: not found
root@yi-5cb1:/# /sbin/imggrabber -r high -w > /tmp/test.jpg
-ash: /sbin/imggrabber: not found

```

roleoroleo, 2020-11-27:
Source the env before sending the command:
```
. /tmp/sd/yi-hack/script/env.sh
```

ayasystems, 2020-11-27:
Thanks @roleoroleo 

root@yi-5cb1:~# imggrabber -r high -w > /tmp/test.jpg
Starting program
Mapping file /dev/shm/fshare_frame_buf, size 1786156, to b6d1c000
Encoding h264 frame
Starting decode
Decode frame
Writing yuv buffer
Cleaning ffmpeg memory
Adding watermark
Encoding jpeg image
Unmapping file /dev/shm/fshare_frame_buf, size 1786156, from b6d1c000

[image]

roleoroleo, 2020-12-27:
@ayasystems 
Did you solve your problem?

jjleto, 2020-12-28:
same problem with r30gb_0.1.2 (also tested r30gb_0.1.3 but with continuous reboot)

imggrabber -r high -w > /tmp/test.jpg
Starting program
Mapping file /dev/shm/fshare_frame_buf, size 1786156, to b6cd0000
Encoding h264 frame
Starting decode
Decode frame
Writing yuv buffer
Cleaning ffmpeg memory
Adding watermark
Encoding jpeg image
Unmapping file /dev/shm/fshare_frame_buf, size 1786156, from b6cd0000

[image]

roleoroleo, 2020-12-28:
Please, make a dump of the file /tmp/iframe.idx with hexdump.
The log shows that imggrabber tries to encode the frame using h264 and not h265.

ayasystems, 2020-12-29:
Hi @roleoroleo 

Here the dump, let me know if need more info

```
root@(none):/tmp# hexdump iframe.idx > iframe.idx.dump
root@(none):/tmp# cat iframe.idx.dump
0000000 7c18 0002 0019 0000 7c41 0002 0008 0000
0000010 7c5f 0002 3c43 0000 a83b 0001 0019 0000
0000020 a864 0001 0008 0000 a882 0001 3c1a 0000
0000030
```

roleoroleo, 2020-12-29:
Which versionvare you using?
iframe.idx should be 64 bytes long.

jjleto, 2020-12-29:
root@dome:/tmp# cat iframe.idx.dump
0000000 2203 001b 0019 0000 222c 001b 0008 0000
0000010 224a 001b 332a 0000 da3a 0004 0019 0000
0000020 da63 0004 0008 0000 da81 0004 3348 0000
0000030
r30gb_0.1.2

roleoroleo, 2020-12-29:
Maybe I found a bug in the OTA upgrade procedure.
Try to reinstall the last version from scratch.

ayasystems, 2020-12-31:
Hi

High resoltion is working now with this firmware:

Firmware Version | 0.1.3
Base Version | 9.0.05.32_202004101552

But seems like this camera has not high resolution. 
[snapshots.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5757138/snapshots.zip)

roleoroleo, 2020-12-31:
Why do you think this cam has not high res?
Probably the quality is not so good but is high res.

ayasystems, 2020-12-31:
May be, the quality is not good..... A old foscam has better quality without hd

jjleto, 2021-01-03:
Tested now with a fresh install of r30gb_0.1.3 on new sd card, hd snapshot when working it’s ok, but the cam have continuous reboot.

roleoroleo, 2021-01-03:
This is a know issue.
It's a low memory condition.
Enable swap file.

Zephael86, 2021-01-15:
I can confirm a successful fresh install using r30gb_0.1.3 to YI Dome Guard (SN: HR303G... DID: HFUSY....).
High resolution snapshot works.
Enables swap file.

thehutgy, 2021-08-05:
> I can confirm a successful fresh install using r30gb_0.1.3 to YI Dome Guard (SN: HR303G... DID: HFUSY....).
> High resolution snapshot works.
> Enables swap file.

I can confirm this too, mine was crashing before, Enabling swap file should work perfectly.

Thanks for this amazing yi hack!

closed 2023-01-08 by stale[bot]