roleoroleo/yi-hack-Allwinner-v2 #34: Cannot use RJ45 Plug LAN

Chluz, 2020-11-22:
Hi, first off thank you very much for these releases. 
I'm using r30gb_0.1.2.tgz for the Yi Dome Guard camera, working pretty well (and I saw that some recent commits will fix the x265 stream, very cool). 

The Dome Guard actually has an RJ45 port at the back, that I prefer using as I do not want streams constantly using up the wifi throughput. 
If a wifi network is configured, the camera doesn't seem to use the LAN RJ45 plug at all. Any way to force the LAN to be used ?

Thanks in advance,

roleoroleo, 2020-11-22:
I didn't know it had a lan.
I will check it.

roleoroleo, 2020-11-26:
What services are not running when you use ethernet?

Chluz, 2020-11-26:
Hi Roleorolea, 

I had turned off Onvif, the Yi interface (to have fully in private mode) and the human AI detection. 
Would you like me to check with Yi interface on ?

roleoroleo, 2020-11-26:
I would like to know if the problem involves only hack features or the yi features too.

Chluz, 2020-11-29:
Hi roleoroleo,

Well after checking in more detail I noticed that :

- I had an issue with the network cable I was using

- the camera switches to rj45 Lan only when the camera is rebooted with lan connected.

Both the things above led me to misdiagnosed the issue, sorry about that

The rstp is working fine from lan, so is mqtt and the Web interface.

Thanks again, I think this issue can be closed

roleoroleo, 2020-11-30:
But I think onvif and wsdd don't work with ethernet.
Could you check?

Chluz, 2020-12-06:
Hi roleoroleo, 

I updated to 0.1.3, and here are my findings

- The only way I can force the camera to reliably connected to LAN (if wifi has been configured) is to block the camera mac address from my wifi 
- Once on Lan, I tested ONVIF with onvifer 14.89 on android. When providing the the camera IP, it starts configuring it, but is unable to download the "media profiles". 
- On the main page of the yi-hack web interface, the LAN ip is not showing, so the links are incorrect
- I was not able to test wsdd, as I'm not familiar with the protocole. Can you suggest a client ?

Thanks again for all this !

roleoroleo, 2020-12-06:
>     * The only way I can force the camera to reliably connected to LAN (if wifi has been configured) is to block the camera mac address from my wifi

This is a weird behavior. My cam has wifi configured, but if I connect the cable, it uses the eth0.
The router shows eth connected and wifi disconnected.

>     * Once on Lan, I tested ONVIF with onvifer 14.89 on android. When providing the the camera IP, it starts configuring it, but is unable to download the "media profiles".

I will check it.

>     * On the main page of the yi-hack web interface, the LAN ip is not showing, so the links are incorrect

This problem is related to the 1st. In my case, links and LAN ip are correct.

>     * I was not able to test wsdd, as I'm not familiar with the protocole. Can you suggest a client ?

Try Onvif Device Manager. When you start it, if it detects the cam automatically, wsdd works.
If you need to add the cam manually, wsdd is not working.

Chluz, 2020-12-13:
Hi Roleoroleo, 

I confirm that the Onvif Device Manager does not find the camera when it is connected to LAN. 
Even entering teh camera IP does not help, so ONVIF does not seem to work either

roleoroleo, 2020-12-13:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/f1aecad11a53fe2ffb9120ee95e6caccb3198ff8

Next release...

Zephael86, 2021-01-18:
I also use r30gb, but with fw. ver. 0.1.3.
Is there anything where I can help?

roleoroleo, 2021-01-18:
@Zephael86 Does your cam work with the ethernet cable?

Zephael86, 2021-01-18:
Not really. Probably because camera connects using wlan0 even if eth0 is plugged in.
I have wlan0 configured. Connected UTP cable before power cable was plugged in and camera connected using wifi.

Executed `udhcpc -i eth0 -b -s /home/app/script/default.script`
Interface got up, obtained IP Address, updated default gateway - but it left the wlan0 with same def gw and metric.

Logged out and logged in using eth0 and ssh.

Following processes have 'wlan0' as parameter:

```
 1154 root      1400 S    /backup/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_supplicant-global -Dnl80211 -iwlan0 -B
 1254 root       980 S    udhcpc -i wlan0 -b -s /home/app/script/default.script
 1563 root      8044 S    onvif_srvd --pid_file /var/run/onvif_srvd.pid --model Yi Hack --manufacturer Yi --firmware_ver 0.1.3 --hardware_id 0000 --serial_num HFUSY36NZ3BGIC200526 --ifs wlan0 --port 80 --scope onvif://www.onvif.org/Pro
 1574 root      2072 S    wsdd --pid_file /var/run/wsdd.pid --if_name wlan0 --type tdn:NetworkVideoTransmitter --xaddr http://%s --scope onvif://www.onvif.org/name/Unknown onvif://www.onvif.org/Profile/Streaming

```

Executed: `ifconfig wlan0 down` in a loop. Did not have the time to check which process brings it up automatically. Even if I killed udhcp for this interface...

Observations:
- Web interface is reachable using eth0
- Web interface displays RTSP and Snaptshot links using the IP of the wlan0. After a while the IP Address part becomes null.
  [image]
- Snapshot works if I replace the IP address in the link for eth0
- WiFi strength indicator shows 0. (This was always 0 for me. Probably this is not related for this investigation).
- Address and Network fields are empty
  [image]
- Onivif application could not find the device.
- Onivif application could connect to the RTSP link when I entered manually.
- Top menu items links in the WebUI are working. Using the Address of eth0
- Snapshot's menu snapshot function works
- After a `reboot` wlan0 was up and eth0 down.

roleoroleo, 2021-01-18:
Probably you can't change the network  without reconfiguring completely the cam.
I configured the cam via ethernet the 1st time and it works correctly.
The process that manages the network is a binary and I don't know how it chooses between eth0 and ra0.

Maybe it's stored in the configuration db?
Try to run this command:
`/mnt/mmc/sonoff-hack/bin/sqlite3 /mnt/mtd/db/ipcsys.db "select * from
 t_sys_param where c_param_name = 'NetWorkType';"`
and check if the output is 0 or 1.

About onvif_srvd and wsdd, you need to change "ONVIF Network Interface" parameter (configuration web page) to select the correct interface.

roleoroleo, 2021-01-18:
The process that manages network configuration is a closed binary (dispatch).
So I don't know how the system chooses the correct network adapter during boot.
And I don't own a r30gb to test it.

About onvif_srvd and wsdd, you can select the interface in the configuration web page.

AndWalu, 2021-07-02:
YI Dome Guard Camera, SN: IFUSY36
IMO the system always chooses the wired connection.
If the cable is connected the Wi-Fi connection is dropped and its IP is released.
I think that to determine a camera IP you have to check at first the IP of a eth0 interface.
If the eth0 IP is set use it,  if not check the IP of a wlan0 and use it.

ifconfig
eth0      Link encap:Ethernet  HWaddr D2:F1:5F:06:2A:36  
          inet addr:10.2.2.2  Bcast:10.2.2.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:306 errors:0 dropped:1 overruns:0 frame:0
          TX packets:554 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:31064 (30.3 KiB)  TX bytes:719581 (702.7 KiB)
          Interrupt:66 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

wlan0     Link encap:Ethernet  HWaddr 54:F1:5F:06:2A:36  
          UP BROADCAST RUNNING MULTICAST  MTU:1200  Metric:1
          RX packets:117 errors:0 dropped:35 overruns:0 frame:0
          TX packets:96 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:29050 (28.3 KiB)  TX bytes:14176 (13.8 KiB)

roleoroleo, 2021-07-03:
Please, check this commit:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/e4045061f4bb97cf040f8f3586b31cab45eb0c36

closed 2023-01-08 by stale[bot]