roleoroleo/yi-hack-Allwinner-v2 #35: MQTT image not published, wrong file path

inbux, 2020-11-23:
Hi,

there is a wrong file path in mqttv4.h for the imggrabber:
 /home/yi-hack/bin/imggrabber

should be 

/tmp/sd/yi-hack/bin/imggrabber

roleoroleo, 2020-11-23:
Ok, thanks.
I will fix it.

mcanteri, 2020-11-25:
@inbux did it worked after that fix ?
I recompiled the source code but don't get the image from the mqtt camera.

inbux, 2020-11-25:
Yes, it works.
But you have to change the patch file, because the mqttv4.h file will be always overwritten with the original file from github every time you recompile all.

roleoroleo, 2020-11-25:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/3dc2ce9d4c3200201d396bcd44f6b88282e0888b

mcanteri, 2020-11-25:
Yes, now I noted it was overwritten when compile was run.
Do you mean the file src/mqtt/dir.patch ?
Never done it manually before.

mcanteri, 2020-11-25:
Great @roleoroleo :-)
Thank you, will try now again.

mcanteri, 2020-11-25:
:+1: now work, great.
@inbux how do you manage to send the image to a notification chat, like telegram ?
I've call the camera.snapshot service on that camera, but I see some problems sending the more recent snapshot, in this way I probably have to fine tune a bit of delay.

marcospg75, 2020-11-29:
> 👍 now work, great.
> @inbux how do you manage to send the image to a notification chat, like telegram ?
> I've call the camera.snapshot service on that camera, but I see some problems sending the more recent snapshot, in this way I probably have to fine tune a bit of delay.

I just make a HTTP Request to the snapshot URL of the yi-cam from Node Red when detect a motion.

mcanteri, 2020-12-02:
So you don't even use this feature (MQTT Image), motion and snapshot were available even before.

closed 2022-06-11 by roleoroleo