roleoroleo/yi-hack-Allwinner-v2 #36: Tested working Yi dome 1080p SN/IFUS

flexiones, 2020-11-27:
Tested on 4 pack yi home 1080p with serial number starting on IFUSXXXXXX.
Working with Qnap QVRpro over RTSP protocol.
[image]
[image]
[image]

roleoroleo, 2020-11-27:
Thank you for your feedback.
Did you try onvif also with qnap?

flexiones, 2020-11-27:
Onvif works as alocating service, But need to re-set as generic camera and rtsp h264.
I Can't setup Onvif, but not spent much time setting up devices. Maybe could try next week, but it's new for me, not understand yet.

closed 2020-12-27 by roleoroleo