roleoroleo/yi-hack-Allwinner-v2 #38: SSH, MQTT settings are not applied when modified via UI

piotr277, 2020-11-28:
Hi, I've got y21ga BFUS with firmware 0.1.2. Since some time everything is working quite well. But recently I've tried to change some settings via web UI and those settings are not aplied after reboot. To be precise I cannot change: MQTT settings (on/off, ip, etc.) and enabling SSH daemon. I've tried also to download config archive via web UI, then modify those values and upload it back. But no effect at all.
Any help, suggestions how to solve it?
Cheers

mcanteri, 2020-11-28:
Explain what is the change you are making, so I can check on mine.

piotr277, 2020-11-30:
Ok, I cannot change MQTT broker IP, I cannot enable MQTT at all. Also I cannot enable SSH access. After camera reboot those values are reverted.

mcanteri, 2020-11-30:
After changing do you save at the end of the page ?

roleoroleo, 2020-11-30:
Clear the cache of your browser.

piotr277, 2020-12-01:
> After changing do you save at the end of the page ?

Yes, of course.

piotr277, 2020-12-01:
> Clear the cache of your browser.

Already tried it, no effect at all.
As I mentioned earlier, even if I upload a config (with my changes) it doesn't work. After reboot changes are not aplied.

mcanteri, 2020-12-02:
Have you tried a different SD card ? Maybe it is the file system corrupted in some way ... (just guessing)

xzaz, 2020-12-03:
> 
> 
> > Clear the cache of your browser.
> 
> Already tried it, no effect at all.
> As I mentioned earlier, even if I upload a config (with my changes) it doesn't work. After reboot changes are not aplied.

Have you check system.conf (/tmp/sd/yi-hack/etc) has been written?

piotr277, 2020-12-09:
@mcanteri interesting idea, I will try to replace it and write about the result

roleoroleo, 2020-12-27:
Did you solve your problem?

piotr277, 2021-01-15:
The problem is not solved yet. I've checked and the SD card is corrupted somehow. So I've changed it to new one, placed a 0.1.3 version. But the www interface is not showing up.. Still investigating this.

piotr277, 2021-01-15:
Finally, it's solved! The source of the problem was indeed in the corrupted SD card. I had to revert to stock fw then install yi-hack v. 0.1.3. And now all the settings are correctly saved via web UI. Thanks for the good suggestion to check the SD card!

closed 2021-01-15 by piotr277