roleoroleo/yi-hack-Allwinner-v2 #39: Big delay with RTSP

Baanin, 2020-11-29:
Using version 0.1.2, I have really big delays (from 2 to 5 sec) using RTSP, with lot of artifacts (on LAN).

SN starting by BFUSY, 
Model suffix y21ga

Firmware Version | 0.1.2
Base Version | 9.0.19.02_202007141512

mcanteri, 2020-11-29:
How is the strength of the signal ?

Baanin, 2020-11-29:
100%

mcanteri, 2020-11-29:
So it is strange you have artefacts. For the delay it is normal if you connect using rtsp (or onvif that it is the save protocol under).

Baanin, 2020-11-30:
Is there a possibility to connect trough another protocol with less delay ?

mAsHERbtm, 2020-12-02:
I have exactly the same. Delays of 5-8 seconds. My model: BFUSY44A4M. It doesn't matter the video quality - HIGH or LOW Video - 100% strong wifi signal

mcanteri, 2020-12-02:
> Is there a possibility to connect trough another protocol with less delay ?

Only using their application is possible to have less delay.

mAsHERbtm, 2020-12-02:
I downloaded the ONVIF Device Manager application on Windows. If I watch live view there, the delays are max 1-1.5 seconds.

my configuration.yaml 

camera:
  - platform: ffmpeg
    name: YiHome
    input: -an -i rtsp://192.168.0.50/ch0_1.h264

maybe it should be set differently?

mcanteri, 2020-12-02:
Mine is only:
  - platform: ffmpeg
    input: rtsp://192.168.1.132/ch0_0.h264
    name: living_room_camera
Extra arguments should go with the specific option like:
    extra_arguments: -rtsp_transport tcp -analyzeduration 15

Try it, don't know if change something, theoretically it shouldn't.

closed 2022-06-11 by roleoroleo