roleoroleo/yi-hack-Allwinner-v2 #40: No mkpasswd - SSH password can't be set.

AndWalu, 2020-12-01:
Version y21ga_0.1.2

mkpasswd  is not available so "system.sh" script can't create PASSWORD_MD5 and setting of password fails.

Workaround:
In a file "/tmp/sd/yi-hack/script/system.sh" change line:
  PASSWORD_MD5="$(echo "${SSH_PASSWORD}" | mkpasswd --method=MD5 --stdin)" 
to:
  PASSWORD_MD5=$SSH_PASSWORD

On any Linux system compute  MD5 hash for your password using "mkpasswd --method=MD5 --stdin"
In file "/tmp/sd/yi-hack/etc/system.conf" set "SSH_PASSWORD" variable to computed MD5 hash.

AndWalu, 2020-12-01:
The busybox includes mkpasswd applet but mkpasswd link is not created, simply copy any busybox  binary to mkpasswd...
  cp  /tmp/sd/yi-hack/usr/bin/tail  /tmp/sd/yi-hack/usr/bin/mkpasswd

roleoroleo, 2020-12-02:
I will update the busybox installer.
Thank you.

closed 2020-12-27 by roleoroleo