roleoroleo/yi-hack-Allwinner-v2 #42: Yi Home BFUSY not working with y21ga_0.1.2

samazw, 2020-12-02:
Hi

opening a new issue as requested.
I have a Yi Home BFUSY with firmware 9.0.19.02_202007141512. 
Unpacked y21ga_0.1.2 to formatted sd card:

This is the serial output without SD card:

.[38]HELLO! BOOT0 is starting Nov 14 2.Û5g.ª›@read erroz
[54]bnard init ok
[56]chip id check OK
[58]DRAM BOOT DRIVE INFO: V0.41
[61]DRAM CLK = 528 MHz
[63]DRAM Type = 2 (2:DDR2,3:DDR3)
[67]DRAMC read ODT  off.
[69]DRAM ODT off.
[71]DRAM SIZE =64 M
[78]DRAM simple test OK.
[80]rtc standby flag is 0x0, super standby flag is 0x0
[86]dram size =64
[87]key press : 
[90]spinor id is: ef 40 17, read cmd: 03
[94]Succeed in reading toc file head.
[97]The size of toc is 4c000.
[150]Entry_name        = optee
[153]Entry_name        = u-boot
[157]Entry_name        = dtb
[161]Jump to secend Boot.
MESSAGE: [0x0] TEE-CORE: OP-TEE version: sun8iw19p1_v0.6.0-12-g97f2688 #1 2019å¹´ 11æœˆ 08æ—¥ æ˜ŸæœŸäº” 08:26:51 UTC arm
ERROR:   [0x0] TEE-CORE:platform_standby_fdt_parse:126: no pmu node
ERROR:   [0x0] TEE-CORE:sunxi_twi_parse_from_dt:84: no pmu node
U-Boot 2018.05-00007-g8b9203a (Jul 14 2020 - 14:49:11 +0800) Allwinner Technology
[00.206]DRAM:  64 MiB
[00.208]Relocation Offset is: f9f94000
[00.220]secure enable bit: 0
[00.222]CPU=816 MHz,PLL6=600 Mhz,AHB=200 Mhz, APB1=100Mhz  MBus=132Mhz
[00.229]gic: sec monitor mode
[00.231]flash init start
[00.233]workmode = 0,storage type = 3
SF: Detected w25q64cv with page size 256 Bytes, erase size 64 KiB, total 8 MiB
[00.250]sunxi flash init ok
[00.253]Loading Environment from SUNXI_FLASH... start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x3e00, len: 0x100
OK
[00.314]try sprite_led_gpio config
xxxxxxxxxxxxxxlijun uboot xiaoyiledinit
[00.322]sprite_led_gpio start
[00.325](weak)update dtb dram start
[00.329]update dtb dram  end
[00.331]update dts
root_partition is rootfs
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
[... 184 lines cut ...]
part_name:rootfs start_offset:0xea0 part_size:1179648 
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
part_name:home start_offset:0x17a0 part_size:3407872 
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
part_name:backup start_offset:0x31a0 part_size:1245184 
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
part_name:env start_offset:0x3b20 part_size:131072 
Nothing updated, start normal boot
Hit any key to stop autoboot:  1 ... 0
=> .þþ.þþ

==========================================================

Result: The camera is remaining in a status with yellow LED, no reaction - meaning no access to it via browser.

Any idea how to proceed?

Here´s the hardware:
[image]

roleoroleo, 2020-12-02:
The log says "** Unrecognized filesystem type **".
Probably your sd card is not correclty detected.
Try to change the sd or try to format FAT32 with a different tool (eg rufus).

Or maybe sd controller damaged?

Gogitt, 2020-12-16:
I have identical problem as samazw, is this problem solved? Issue not closed though...

roleoroleo, 2020-12-27:
The problem is not related to the hack.
the camera has many compatibility problems with SD card and FAT32 file system.

samazw, 2021-01-11:
sd card was fine, also FAT32 System. But there´s a camera series out setting a killbit (tried four cameras, each of a set of two cameras).
Now returned to amazon, got two new ones from Aliexpress and they work both with same sd card.
The label outside the Ali camera´s are identical to amazon´s ones (both versions Yi Home BFUSY with firmware 9.0.19.02_202007141512.). I will open a case of the Ali ones soon to see if there´s something different inside.

Thank roleorolo et al for this great job :-)

closed 2021-01-11 by samazw