roleoroleo/yi-hack-Allwinner-v2 #45: Web-Service to get Status of Services

dcbo, 2020-12-05:
Hi There,

I'm not sure, if I missed something, but I didn't find the answer to the following 

**Question:**
With the Web-Service a can activate and deactivate the mp4record service like this:
```
http://IP:8080/cgi-bin/service.sh?name=mp4record&action=stop
http://IP:8080/cgi-bin/service.sh?name=mp4record&action=start
```
But is there a way to get the actual state of the Service. 

**Background:**
I want to use this as a home surveillance, but only if I am not at home.
With action=stop and action=start I am able to control this, but 
I want also a Status Light, which tells me, if the Alarm is active.
The only way i found is SSH to the cam and issue `ps | grep mp4record`

Sure i can handle this in my backend, which controls the Web-Service, 
but it would be more elegant when I could grab the real status.

If this is not possible i would like if this is taken as a feature request.

- Web-Service to get Status of Services _or_
- Web-Service to get Output of `ps`

roleoroleo, 2020-12-05:
I will add it.

roleoroleo, 2020-12-05:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/63ae39b3b2778b94e1e7a0001af9c90341be4da6

dcbo, 2020-12-06:
You are my hero!
If I want to test it, I have to compile it, right.

During compilation I got the following Error: (Debian under WSL)
after issuing `sudo ./scripts/compile.sh`
```
checking for bison version... 1.21.9-1
configure: error: You need bison version 3.0 or greater, or use --disable-maintainer-mode.
```

I solved the problem by installing Bison 3.6
```
wget ftp://ftp.gnu.org/gnu/bison/bison-3.6.tar.xz
tar xf bison-3.6.tar.xz
cd bison-3.6
./configure
make
sudo make install
```
Hope that was right.

But now, how do I install the new Version, I think OTA is not possible.
Am I right, that I have to perform the same steps which i did at the first hack?
- Switch Cam off
- get SD-Card out of Camera
- copy Files from `y21ga_0.1.3.tgz` to root of SD-Card 
- insert SD-Card to Camera
- Switch Cam on

dcbo, 2020-12-06:
One more Question:
I compared the `hack y21ga_0.1.3.tgz` with the one i compiled and noticed, 
that the complete Directory `yi-hack\www\` is missing in my self compiled `y21ga_0.1.3.tgz`

Is this OK?

I attached my self compiled `y21ga_0.1.3.tgz`, packed in a ZIP-File, 
because github does not support tgz.

[y21ga_0.1.3.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5648155/y21ga_0.1.3.zip)

roleoroleo, 2020-12-06:
www directory is needed.
Probably there is a problem during the build.
But it's not necessary to rebuild all in this case.
Download the service.sh file and overwrite your current file.

dcbo, 2020-12-06:
Perfect.
I did it remotely over ssh:
`vi /tmp/sd/yi-hack/www/cgi-bin/service.sh`

Thanks again!

closed 2020-12-06 by dcbo