roleoroleo/yi-hack-Allwinner-v2 #47: Cant flash my yi 1080p home camera BFUS

chechilll, 2020-12-08:
Cant flash my yi 1080p home camera BFUS.
I open the camera and got Y29GA_MB_V2.1 2020/04/07 with firmare 9.0.19.02_202007141512
Chipset QG2101A L1026AA
I tryed with diferents SD's, (128Mb, 2Gb, 8Gb, 16Gb and 32Gb) all in fat32 format.
Put the files downloaded from --> https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases --> file y21ga_0.1.2
I decompress the folder and put in the SD. Put the SD and power on the camera.

https://user-images.githubusercontent.com/57603613/99558616-3554a280-29c4-11eb-8a97-af1f3f5f1a65.png

I've tried several times. Never work, always boot normal. Got ERR_CONNECTION_REFUSED if i put the http://IP_CAM:8080
SSh conection refused also
After flash intend, the content of the SD looks like this

https://user-images.githubusercontent.com/57603613/99559353-fd019400-29c4-11eb-9c9c-6ae03d3b9d14.png

It's not a connection problem. If the hack works, i've the followinf problems.
I'm reaching the IP address with ping.

https://user-images.githubusercontent.com/57603613/99664968-46a3ba80-2a69-11eb-9eda-533991b243de.png

I'm not able to connect with no one browser, as you can see it says connection refused as i said.

https://user-images.githubusercontent.com/57603613/99665095-72bf3b80-2a69-11eb-9dc0-a4bb32b1ef2d.png

Also cant connect with SSH to port 22, connection refused

https://user-images.githubusercontent.com/57603613/99665158-8a96bf80-2a69-11eb-91a6-c0812a569372.png

i've tried in two diferent wifi's (SSID in 2.4Gz)

Tryed to connecti with 2 pc's and 2 mobiles. Same problem. Connection refused

https://user-images.githubusercontent.com/57603613/99687594-1ff37d00-2a85-11eb-839c-fd9ca2f7aaf3.png

roleoroleo, 2020-12-08:
Factory.done means that the hack works.
I don't know why the services don't start.
Try to decompress the archive with a linux machine.

chechilll, 2020-12-08:
I used a w10 with 7 zip software to decompress it. Can you pass a MD5 checksum better?

roleoroleo, 2020-12-08:
7zip is ok.
I don't know.
Make a backup and send it to me.

chechilll, 2020-12-09:
You mean a dump with this?
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Dump-your-backup-firmware-(SD-card)
How can i send you? PM?

roleoroleo, 2020-12-09:
Yes, the procedure described in the wiki.
Get the email address from my paypal link.

roleoroleo, 2020-12-11:
The reason why the hack doesn't work is that your init.sh contains telnetd command.
But it doesn't contain the link to the hack.

If you check the sdhack you can see these lines:
```
### Check if hack is present
echo "### Hacking"
if [ `grep telnetd /backup/init.sh | grep -c ^` -gt 0 ]; then
    echo "Hack already applied"
else
    echo "Applying hack"
...
```

The script finds telnetd and exits without applying the hack.
But why do you have telnetd command? Did you apply a previous beta hack?

Solution: edit init.sh removing telnetd command and restart the hack.

chechilll, 2020-12-11:
I fint the lines in factory\config.sh
Remove this lines?

echo "### Hacking"
if [ `grep telnetd /backup/init.sh | grep -c ^` -gt 0 ]; then
    echo "Hack already applied"
else

Flugdino, 2020-12-14:
Have quite the same issue (identical SW and serial) same telnetd-line in the init.sh, but no Factory.done file. If I understood the lines right it doesn't help to wipe out the telnetd line only cause the new init may be/is already written to the cam? so it ends in a circle?

roleoroleo, 2020-12-14:
If Factory is not renamed in Factory.done the hack procedure is not processed.
Try a different SD card or reformat it with a different tool (fat32).

chechilll, 2020-12-14:
> The reason why the hack doesn't work is that your init.sh contains telnetd command.
> But it doesn't contain the link to the hack.
>
> If you check the sdhack you can see these lines:
>
> ```
> ### Check if [...]

Thanks roleoroleo,
It works now!! 
I erase the line **/usr/sbin/telnetd** & in file \newbackup\init.sh.
What this do if missing? can't use telnet? or only for the hack?
I need erase in new furete hack versions?
Kind Regards!!!

roleoroleo, 2020-12-14:
When the hack runs, it add telnetd to the init.sh. You can use it.
No, you don't need to erase the line in the future,

chechilll, 2020-12-14:
Many thanks for all your work and support.
Kind Regards

closed 2020-12-14 by chechilll