roleoroleo/yi-hack-Allwinner-v2 #48: Firmware won't flash on 1080 BFUS?

xGoJox, 2020-12-08:
Hello,

Searching on the issues in YiHack v4 repo, i ended up here, so i guess this is the right place>>looks like there are 3 different AI+, how can i make sure which i got?! In any case, i tried with ALL of the version, i just extract files on FAT32 sdcard, insert it in the YI and then finally plug its USB to power on.. BUT it just will never flashes : it just starts normally > orange led, then blue = connected. What am i doing wrong? Hope i'll get help here. 

Thank :)

roleoroleo, 2020-12-08:
Fimrware version?
Home, dome, ...?

xGoJox, 2020-12-08:
It's Home & i just checked firmware's 9.019.02; i'm asked to upgrade to 1.8.7.0F, which is older apparently?!

roleoroleo, 2020-12-09:
9.0.19 sounds good.
Did you try if the hack is working?
Try to connect to http://IP_CAM:8080

xGoJox, 2020-12-09:
Nothing happens trying to connect. Not surprising since, as said, firmware never flashed!

roleoroleo, 2020-12-09:
Try to make a backup with the procedure explained in the wiki and send it to me.
I will check if the cam is compatible.

roleoroleo, 2020-12-09:
There isn't another way.

lukaszpluku, 2020-12-09:
Try to do as i do i have exactly that same cam and firmware versione. I flash by originale firmware. Next download Allwiner 2, pepare 16 GB card ad but format not in windows as usually-but some other windows app. Next put Allwiner 2 on SD card, turn of cam, put in SD card. Turn on cam and wait. If nothing heard (as i have) repeat this steps.

closed 2022-06-11 by roleoroleo