roleoroleo/yi-hack-Allwinner-v2 #49: Home Assistant Integration

mAsHERbtm, 2020-12-09:
I have question about integration with Home Assistant

1. How to properly add a camera to the Home Assistant so that all sensors are visible (motion detection, human detection, baby crying).
2. How add a camera so that the motion sensor is displayed with camera view.

I was able to add a camera to HA but I do not have a motion sensor in view.
After adding sensors in the binary sensor, only the motion sensor works (all sensors from the application level are enabled: http://ip_camera:8080/?page=camera_settings). 
Not working Human Detection and Baby Crying.

my file: **configuration.yaml**

```
stream:
ffmpeg:
camera:
  - platform: ffmpeg
    name: yi_hack_v4
    input: -rtsp_transport tcp -i rtsp://ip_camera/ch0_0.h264

binary_sensor:
  - platform: mqtt
    state_topic: "yicam/motion_detection"
    name: "yicam_motion_sensor"
    payload_on: "motion_start"
    payload_off: "motion_stop"
    device_class: motion
  - platform: mqtt
    state_topic: "yicam/ai_human_detection"
    name: "yicam_human_sensor"
    payload_on: "human_start"
    payload_off: "human_stop"
    device_class: motion
  - platform: mqtt
    state_topic: "yicam/baby_crying"
    name: "yicam_baby_sensor"
    payload_on: "crying"
    device_class: sound
```

roleoroleo, 2020-12-09:
See the wiki.

timknowlden, 2020-12-20:
Unable to auto discover mqtt?

roleoroleo, 2020-12-20:
Auto discovery isn't implemented.

timknowlden, 2020-12-21:
No problem, used the above config.YAML but however not picking up motion, human or baby crying

timknowlden, 2020-12-21:
I have attempted with endless amounts of config to get this integrated, but the sensors pick up nothing in HA. I have followed the wiki word for word.

roleoroleo, 2020-12-21:
Try to disable human detection.
There is a know problem with this feature..
If you enable it you don't receive events (even in the origina yi app).

timknowlden, 2020-12-21:
Aha, finally, given me a headache! hopefully will be fixed in future firmware then. thank you.

roleoroleo, 2020-12-21:
I don't know if Yi will fix this problem.
I think not.

closed 2022-06-11 by roleoroleo