roleoroleo/yi-hack-Allwinner-v2 #50: MQTT status rapidly change

korasinski, 2020-12-09:
Hi!

It seems that MQTT integration pings all the time messages about status.
Status is getting online state and around 1,5 seconds later it becomes offline.

[image]

Firmware Version: 0.1.3
Base Version: 9.0.19.02_202007141512
Model Suffix: y21ga
Serial Number: BFUSY******

roleoroleo, 2020-12-09:
This is not the correct behavior.
Check your connection because "offline" message could be sent from the broker when the connection goes down.
If the cam loses the connection to the broker, it will send "offline" message.
I checked on my cam and I have no problems.

korasinski, 2020-12-09:
Yes! You were right, something was not configured. I restored yi-hack settings to the default one, rebotted the cam and now everything works! 

[image]

Maybe someone will have simmilar issue so this may be the fix.

closed 2020-12-09 by korasinski