roleoroleo/yi-hack-Allwinner-v2 #51: Auto video cleaning is not working

huntz, 2020-12-10:
The "Auto video cleaning" option is not working.

Right now I have 6% of free space on my SD, but the "Percentage of free space to keep in the SD" is set to 50.

I'm running the y21ga model.

roleoroleo, 2020-12-10:
Try this commit:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/4f48554fd3ac516849e696f76a472e007e005015

huntz, 2020-12-11:
Hi,
running the command `/tmp/sd/yi-hack/script/clean_records.sh 50` from ssh after the patch, now works. The script deletes the old recordings as expected.

I suppose also the hourly cron job will work.

Thank you!

huntz, 2020-12-11:
Nope, it's not working :-(

I can see the cronjob file correctly created ( /var/spool/cron/crontabs/root ) but the crond process is missing.

As I can see, the crond executable must be in the directory /tmp/sd/yi-hack/usr/sbin/ but it's missing (i can see only _httpd_ and _ntpd_).

So, the clean_records.sh is working if run manually from shell, but the hourly cronjob is not executed by crond (because the daemon is missing)

roleoroleo, 2020-12-12:
Fixed.
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/0070af5fa80c1857ab2cd4ee5d7c14c66a860226
You can add crond to your cam copying busybox binary.
`cp /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/crond`

huntz, 2020-12-12:
Sorry, but the busybox is missing

`# cp /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/crond`
`cp: can't stat '/tmp/sd/yi-hack/bin/busybox': No such file or directory`

roleoroleo, 2020-12-12:
You are right.
`cp /tmp/sd/yi-hack/bin/base64 /tmp/sd/yi-hack/usr/sbin/crond`

huntz, 2020-12-12:
Ok. Now it's working.

I think we can close this issue...

Thank you!

closed 2020-12-12 by huntz