roleoroleo/yi-hack-Allwinner-v2 #52: BFUS 9.0.19* - brick

mates21, 2020-12-15:
Can I unbrick cam if i don't have backup files? :(
when i updated to 0.1.3 the camera only has yellow light. Same without sd card.

roleoroleo, 2020-12-15:
Very strange this brick...
But you need a backup to unbrick your cam.
Another backup of the same model is enough.
Maybe another user could share it with you.

mcanteri, 2020-12-15:
Hi @mates21 
I shared my backup here, let me know if you can download it (first time I share using MEGA):
https://mega.nz/folder/sCwXFIiJ#hs9HPcYackYvCZ0ThfKImA

mates21, 2020-12-15:
Thank you very much @mcanteri ! I've already downloaded and I'll try to fix the camera tomorrow
I must open my cam and try this https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Unbrick-the-cam ? is there any other way?

mcanteri, 2020-12-15:
Well, I never had to unbrick the camera so I suppose that is the reference guide.
But are you sure it is briked ? Have you tried to take out the sd card and reset the camera ? It should factory reset it and the original software should back to work. The hack is not writing anything on the internal memory of the camera, so it is hard to make damages, but maybe could happen.

mates21, 2020-12-16:
Without sd card, same situation - yellow light only :(
reset also did not help. after turning on the power, immediately yellow paw and that's it

maocas, 2020-12-16:
I have verified that in my camera the yellow light stays on and hanging when I start the camera with a lower power source. With the original power supply or higher power it starts correctly

roleoroleo, 2020-12-17:
> I have verified that in my camera the yellow light stays on and hanging when I start the camera with a lower power source. With the original power supply or higher power it starts correctly

I had the same problem with a similar model.

> I must open my cam and try this https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Unbrick-the-cam ? is there any other way?

There is no other way, but the wiki page is tested, another user used it.

roleoroleo, 2020-12-27:
@mates21 
Did you restore the cam?

mates21, 2020-12-30:
@maocas it's not the charger fault - I checked on a few, also on the original one.
@roleoroleo unfortunately no :( for me the instruction is too difficult - I'm an amateur in such things :(

trtracer, 2020-12-31:
Had the same problem. Nothing worked so I connected to the serial port of the Cam to see more details. Turned out, the Backup-Partition was full and the init.sh was truncated. I restored the Backup-Partition (thanks to @mcanteri !) and now it works (with 0.1.2) again. Didn't try upgrading to 0.1.3 yet.

roleoroleo, 2020-12-31:
Unfortunately this is a problem.
Backup partition is almost full and the jffs driver doesn't handle very well the free space whel it's low.

speendo, 2021-07-21:
It seems I ran into the same problem.

Lucky me also ripped off the RX solder point. Is there any other place where I can solder the RX connection?

speendo, 2021-07-21:
> It seems I ran into the same problem.
> 
> Lucky me also ripped off the RX solder point. Is there any other place where I can solder the RX connection?

Lucky me managed to get the camera working again. Actually I didn't really do anything. When I turned it on again today, it was back to normal. Maybe it just didn't like the SD card or the power supply I was using yesterday. Or maybe I just had to leave it switched off for some time in order to get it working again...

However, now the RX solder point is ripped off so I don't have a chance to repair it, in case it breaks in the future. Therefore I would still be interested if there is another place to solder the RX connection.

roleoroleo, 2021-07-22:
> However, now the RX solder point is ripped off so I don't have a chance to repair it, in case it breaks in the future. Therefore I would still be interested if there is another place to solder the RX connection.

To find another place we need a high res picture of the pcb.

speendo, 2021-07-27:
I see. I cannot provide that at the moment. Thanks anyway!

zalewma, 2022-11-29:
Hi Guys, Link is not available anymore... Can you share it again please...
I've run into the same issue ...

mcanteri, 2022-11-29:
> Hi Guys, Link with backup is not available anymore... Can you share it again please... I've run into the same issue ...

Here is the backup data for the y21ga model
https://mega.nz/folder/lKo1GSgT#LGi8dJt7DdmUZLWXmEjv1g

closed 2023-03-29 by stale[bot]