roleoroleo/yi-hack-Allwinner-v2 #53: Stream on Home Assistant side, ver. 0.1.3

mcanteri, 2020-12-15:
I recently updated the firmware to 0.1.3 using the feature on maintenance tab and worked like a charm, but then I started to see some issues on HA side and in the log file I see:
`2020-12-15 22:26:00 ERROR (stream_worker) [homeassistant.components.stream.worker] Error demuxing stream: 
2020-12-15 22:26:00 DEBUG (stream_worker) [homeassistant.components.stream.worker] Restarting stream worker in 10 seconds: rtsp://192.168.1.111/live/ch00_1
2020-12-15 22:36:33 ERROR (stream_worker) [homeassistant.components.stream.worker] Error demuxing stream: 
2020-12-15 22:36:33 DEBUG (stream_worker) [homeassistant.components.stream.worker] Restarting stream worker in 10 seconds: rtsp://192.168.1.132/ch0_0.h264
2020-12-15 22:41:06 ERROR (stream_worker) [homeassistant.components.stream.worker] Error demuxing stream: 
2020-12-15 22:41:06 DEBUG (stream_worker) [homeassistant.components.stream.worker] Restarting stream worker in 10 seconds: rtsp://192.168.1.111/live/ch00_1
2020-12-15 22:41:16 ERROR (stream_worker) [homeassistant.components.stream.worker] Error demuxing stream: 
2020-12-15 22:41:16 DEBUG (stream_worker) [homeassistant.components.stream.worker] Restarting stream worker in 20 seconds: rtsp://192.168.1.132/ch0_0.h264
2020-12-15 22:46:29 ERROR (stream_worker) [homeassistant.components.stream.worker] Error demuxing stream: 
2020-12-15 22:46:29 DEBUG (stream_worker) [homeassistant.components.stream.worker] Restarting stream worker in 30 seconds: rtsp://192.168.1.132/ch0_0.h264
`
I enabled DEBUG but don't get much more information. Could it be because I enabled the audio in the stream that before I had it muted ? Anyone else experiencing this ?

mcanteri, 2020-12-15:
Found casually this in log after a reboot:
`2020-12-15 23:49:50 ERROR (stream_worker) [libav.h264] missing picture in access unit with size 8652
2020-12-15 23:49:50 ERROR (stream_worker) [libav.h264] No start code is found.
2020-12-15 23:49:50 ERROR (stream_worker) [libav.h264] Error splitting the input into NAL units.`

roleoroleo, 2020-12-16:
Do you noticed this behavior in 0.1.2 also?

mcanteri, 2020-12-16:
No, before wasn't showing this error but I can't guarantee is up to the camera because even HA received updates.
Maybe someone else is experiencing same behaviour ?

mcanteri, 2020-12-16:
Today I found a core file in the record folder, it says:
file core
core: ELF 32-bit LSB core file, ARM, version 1 (SYSV), SVR4-style, from '/tmp/sd/yi-hack/bin/imggrabber -r high -w', real uid: 0, effective uid: 0, real gid: 0, effective gid: 0, execfn: '/tmp/sd/yi-hack/bin/imggrabber', platform: 'v7l'
with time of this afternoon. This never happen before.

roleoroleo, 2020-12-17:
Ok, I will check the differences between 0.1.2 and 0.1.3.

roleoroleo, 2020-12-17:
imggrabber is the program that makes snapshots. It's not related to the rtsp stream but I will check if there is a sefgault in the code.
About rtsp, try this new release, I removed an experimental section added in 0.1.3:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5707809/rRTSPServer.gz)

mcanteri, 2020-12-17:
The rRTSPServer now is working correctly, no more errors, thank you 👍

timknowlden, 2020-12-18:
Cannot get audio to work in HA, is it possible?

mcanteri, 2020-12-21:
No, the sound never worked on HA to me and I think is an issue on the HA side. Probably should open an issue there.

closed 2020-12-27 by roleoroleo