roleoroleo/yi-hack-Allwinner-v2 #54: Unable to flash on IFUS (Yi Home 1080p) FW 9.0.19.*

changetotower, 2020-12-16:
Hey there

Just tried to flash the rooted firmware on my Yi Home 1080p cam (IFUS, FW 9.0.19.*), but unfortunately without success. Followed the procedure exactly as described. Tried 4GB and 8GB SD-card, formatting with Windows and also from within the Yi Home app. Nothing helped, camera is starting normally, i.e. steady orange, blinking blue, steady blue. Hence, can't connect to http://ip-cam:8080 either.

Any idea?

Many thanks in advance.

flexiones, 2020-12-16:
Try to format inside camera with yi-home phone app.

I've success on first try with cheap micro sd 32gb Netac.

[image]

Gogitt, 2020-12-16:
how to format sd using app?

Gogitt, 2020-12-16:
And what have you done after that?

changetotower, 2020-12-17:
> Try to format inside camera with yi-home phone app.
> 
> I've success on first try with cheap micro sd 32gb Netac.
> 
> [image]
>  

Yes, I tried that as well, as mentioned in my post. But without luck...

changetotower, 2020-12-17:
Ok, got it working:

Most important step, for me at least, is that you add the port :8080 behind your Cam IP when trying to access the hack from a browser. 

That did the trick.

closed 2020-12-17 by changetotower

MarkoWlodarko, 2021-03-18:
Hi @changetotower I'm facing same problem and dont get your magic trick that solved the problem. Could you explain it a little bit.
What doeas it mean 

> add the port :8080 behind your Cam IP 

What is the diffrence in putting in browser simply http://cam-ip:8080 ?