roleoroleo/yi-hack-Allwinner-v2 #55: Video folder is not deleted after FTP video upload

jorgedeand, 2020-12-17:
Hello,

The issue is related to this camera:
Yi 1080p Home BFUS - Version: 9.0.19.02_202007141512
yi-hack-Allwinner-v2 - 0.1.3

I have the following configuration:
-Disable Cloud: ON
-Recodring without Cloud: ON
-Save a video when a motion is detected: ON
-FTP Upload: ON
-FTP Delete: ON
-Create Directory tree: ON

When a motion is detected the camera stores the video in a folder: Example:  /2020Y12M17D11H/51M44S16.mp4
getlastrecordedvideo.sh outputs this: 2020Y12M17D11H/51M44S16.mp4

The video is sent through FTP and deleted from the folder, but the folder remains in the camera.
In the section Motion Events, under Select Time there is still a directory 2020Y12M17D11H.
When I open it it says "No events in this folder"

getlastrecordedvideo.sh outputs this: 2020Y12M17D11H/

eventsdir.sh outputs this: 
{"records":[{
"datetime":"Date: 2020-12-17 Time: 11:00",
"dirname":"2020Y12M17D11H"
}]}

I have another camera, with yi-hack-Allwinner (first version) where the folder is erased successfully after each video.
The other camera:
Yi 1080p Home BFUS - Version: 8.2.0.0A_201912270941
yi-hack-Allwinner - 0.2.2
Same configuration

roleoroleo, 2020-12-17:
I already fixed this problem: https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/917aec9b8f7bda56ed5a1d6c367c1183b6f47e85
Are you sure you installed the last version?
Check if your ftppush.sh file is updated with the master.

jorgedeand, 2020-12-17:
Yes, the last version (0.1.3) is installed.
I also checked ftppush.sh file in the SD card, it is updated.

roleoroleo, 2020-12-17:
Fixed: https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/3f858174f211cf64faa13e3e5f4b6d3ddbf5b458

tjuanma, 2021-01-27:
I was having the same problem and i've just added manually a "rm -rf" line in ftppush.sh file ;)

jorgedeand, 2021-02-24:
I confirm it is fixed in 0.1.4. It can be closed

closed 2021-02-24 by jorgedeand