roleoroleo/yi-hack-Allwinner-v2 #57: Yi Camera BFUS Firmware version 9.0.19.02_202007141512

r4zi3lrrr, 2020-12-19:
Hi Guys,

Extracted to SD as per below screenshot, put the SD card into the Yi camera - left it for several minutes but not getting anything else, just a amber light.

Using yi-hack-Allwinner-v2 FYI

Any ideas, please?

[image]

r4zi3lrrr, 2020-12-19:
Just to add if I remove the SD card the camera boots up normally.

roleoroleo, 2020-12-19:
The content of the sd card you posted is before or after to try the hack?

r4zi3lrrr, 2020-12-19:
After the hack

roleoroleo, 2020-12-19:
If Factory is not renamed Factory.done the hack didn't start.
Try to use a different sd card or try to reformat the card with another software (eg. Rufus).

r4zi3lrrr, 2020-12-19:
Tried another SD card and formatted it using MiniTool Partition Wizard 11

Unfortunately still not working just stuck on static amber light - waited for 5 mins.

If the hack works should the color of the LED change?

Should the folder rename to factory.done or a file?

[image]

This is what I have in the factory folder:

[image]

r4zi3lrrr, 2020-12-23:
Tried another SD card (formatted to FAT 32) and plugged in directly into the mains still stuck on amber light and won't go any further.

r4zi3lrrr, 2020-12-23:
It does say AI+ on the back of the camera - could this be model of camera be too new perhaps?  Maybe why this firmware doesn't work?

r4zi3lrrr, 2020-12-23:
Issue finally resolved due to a good guy from Hotukdeals:- jezcooke

The problem was due to the way Winzip was extracting the TAR file.

Link to the fix that helped me: https://support.esri.com/en/technical-article/000002710

closed 2020-12-23 by r4zi3lrrr