roleoroleo/yi-hack-Allwinner-v2 #58: Why it has to connect to Xiaomi app through Internet?

occoder, 2020-12-22:
I'm wondering why this hack does not workaround the necessity of connecting to Xiaomi app through Internet?
Can we just configure the WIFI router and camera pair using local configuration file in the SD card? Such would be much more straightforward and less 'region locked' hassle involved.

roleoroleo, 2020-12-22:
See the wiki: https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Change-WiFi-credentials

occoder, 2020-12-23:
Thank you, @roleoroleo 
I've managed to connect to my WIFI without Internet following the guidance you gave above.
Then I came to http://IP-CAM:8080 to try out the listed features and found following features work

- rtsp://IP-CAM/ch0_2.h264 (only audio)
- PTZ movement control

But following features do not work

- rtsp://IP-CAM/ch0_0.h264 (high res)
- rtsp://IP-CAM/ch0_1.h264 (low res)
- http://IP-CAM:8080/cgi-bin/snapshot.sh?res=high&watermark=yes
- http://IP-CAM:8080/cgi-bin/snapshot.sh?res=low&watermark=yes

The oddest thing is I put my camera just atop my WIFI router, but the "WiFi Strength" indicator shows me 0%.
Could you help advise more on how to fix these issues? Thanks in advance.

closed 2020-12-23 by occoder