roleoroleo/yi-hack-Allwinner-v2 #59: Yi 1080p home BFUS Firmware version 9.0.19.02  Not Loading

Kipper641, 2020-12-22:
Hello and thanks for the great work.

v 1.2 loads and streams.
v 1.3 copied to card, camera plugged in, wait a minute camera goes to blue light.
However has booted to Yi firmware and not the hack, checking SD Card shows  Factory,done
Not discoverable via Onvif manager or http://IP-CAM:8080, however works with the YI home app so appears hack not loading ?
Have tried formatted with Win and formatted within camera.
Tried to create backup but only a temp folder created.

    Many Thanks

r4zi3lrrr, 2020-12-22:
Hi,

I'm getting roughly the same but my camera won't even load up when I had the SD card in the slot- stay on steady amber light.

Unfortunately Firmware definitely not updated.

roleoroleo, 2020-12-22:
It's a frequent issue.
Try to create again the sd.

r4zi3lrrr, 2020-12-22:
I'm extracting y21ga_0.1.3.tgz directly onto my SD card - does that sound right?

I have this camera:

Firmware 9.0.19.02_202007141512
SN: BYUSy44

r4zi3lrrr, 2020-12-22:
Tried another SD card and formatted it using MiniTool Partition Wizard 11

Unfortunately still not working just stuck on static amber light - waited for 5 mins.

If the hack works should the color on LED of the Yi camera change?

Should the folder rename to factory.done or a file - if the hack works?

image

This is what I have in the factory folder:

image

timmynana, 2020-12-22:
I have the same situation, i've tried many different sd cards and non work. I even tried one sd card from a previous generation camera known to be picked up (yi hack 4 compatible). Is 16gb  really the maximum? I've tried to partition the sd card to only 16gb as the only ones I have are larger. 

I did notice these cameras are a lot more sensitive to the power and depending on the power source some do not boot to the point of flashing.

roleoroleo, 2020-12-22:
Did you solve changing power supply?

timmynana, 2020-12-22:
> Did you solve changing power supply?

I can get it to boot with a different power supply, but no luck with flashing. As the OP stated, it turns yellow when powering on and as soon as the shutter clicks to start the flashing it goes straight to the blue flashing light signalling it is connecting to wifi rather than flashing. Once fully loaded the normal http links do not access the camera.

Thanks for your help!

roleoroleo, 2020-12-22:
Arr you using port 8080?

timmynana, 2020-12-22:
> Arr you using port 8080?

Not being overly computer literate, I have tried the link at #4 of your instructions (6. Check the hack opening the web interface http://IP-CAM:8080). I've also tried just using the camera's ip address.

ArclonT, 2020-12-23:
Hi. Same for me. Same firmware 9.0.19, SD card 16GB, not working. connecting via mobile app working...

roleoroleo, 2020-12-23:
Make a dump of the flash and share it with me.
I will check the problem.
See the wiki.

Kipper641, 2020-12-23:
Have created a backup, how do i share it with you ?

   Many Thanks

Kipper641, 2020-12-24:
All now working :)
Tried different cables , formatting, used winzip instead of winrar.

Ended up with original cable formatted in laptop to fat32 and extracted with winrar.
http://ip-cam:8080/ still found nothing so checked cameras ip address in app and used that with :8080 on the end.
It seems that even after the light has tuned blue it takes a while to settle down, so just gave it a couple of minutes.

Many Thanks and Happy Christmas.

Copro, 2021-01-05:
Awesome worked with base version 9.0.19.02_202007141512 and latest version 0.1.3
Serial Number / DID: BFUS
Model: YI 1080p Home Camera (AI+)

closed 2021-01-16 by Kipper641