roleoroleo/yi-hack-Allwinner-v2 #62: LOW Res rtsp Stream not avilable

mzander, 2021-01-05:
Hi,

i recently flashed the firmware onto my BFUS 9.x Yi Home Camera and it worked great. Thanks for that.

The only issue I got is that I cant access the low resultion stream.
High Res and Audio Only is working.

The log of VLC says this:

```
main debug: VLC media player - 3.0.6 Vetinari
main debug: Copyright © 1996-2018 the VideoLAN team
main debug: revision 3.0.6-0-g5803e85f73
main debug: configured with /Users/d-fu/vlc-3.0/extras/package/macosx/../../../configure  '--prefix=/Users/d-fu/vlc-3.0/build/vlc_install_dir' '--enable-macosx' '--enable-merge-ffmpeg' '--enable-osx-notifications' '--enable-faad' '--enable-flac' '--enable-theora' '--enable-shout' '--enable-ncurses' '--enable-twolame' '--enable-realrtsp' '--enable-libass' '--enable-macosx-qtkit' '--enable-macosx-avfoundation' '--disable-skins2' '--disable-xcb' '--disable-caca' '--disable-pulse' '--disable-sdl-image' '--disable-vnc' '--build=x86_64-apple-darwin17' '--host=x86_64-apple-darwin17' '--with-macosx-version-min=10.7' '--with-macosx-sdk=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk' '--with-breakpad=https://mac.crashes.videolan.org' 'build_alias=x86_64-apple-darwin17' 'host_alias=x86_64-apple-darwin17' 'CC=/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang' 'CFLAGS=-g' 'LDFLAGS=' 'CXX=/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++' 'CXXFLAGS=-g' 'OBJC=/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang' 'OBJCFLAGS=-g'
main debug: processing request item: rtsp://192.168.1.204/ch0_1.h264, node: Playlist, skip: 0
main debug: resyncing on rtsp://192.168.1.204/ch0_1.h264
main debug: rtsp://192.168.1.204/ch0_1.h264 is at 13
main debug: starting playback of new item
main debug: resyncing on rtsp://192.168.1.204/ch0_1.h264
main debug: rtsp://192.168.1.204/ch0_1.h264 is at 13
main debug: creating new input thread
main debug: Creating an input for 'rtsp://192.168.1.204/ch0_1.h264'
main debug: requesting art for new input thread
main debug: using timeshift granularity of 50 MiB
main debug: using default timeshift path
main debug: `rtsp://192.168.1.204/ch0_1.h264' gives access `rtsp' demux `any' path `192.168.1.204/ch0_1.h264'
main debug: creating demux: access='rtsp' demux='any' location='192.168.1.204/ch0_1.h264' file='(null)'
main debug: looking for access_demux module matching "rtsp": 14 candidates
main debug: looking for meta fetcher module matching "any": 1 candidates
lua debug: Trying Lua scripts in /Users/zander/Library/Application Support/org.videolan.vlc/lua/meta/fetcher
live555 debug: version 2016.11.28
lua debug: Trying Lua scripts in /Applications/VLC.app/Contents/MacOS/lua/meta/fetcher
lua debug: Trying Lua scripts in /Applications/VLC.app/Contents/MacOS/share/lua/meta/fetcher
main debug: no meta fetcher modules matched
main debug: looking for art finder module matching "any": 2 candidates
lua debug: Trying Lua scripts in /Users/zander/Library/Application Support/org.videolan.vlc/lua/meta/art
lua debug: Trying Lua scripts in /Applications/VLC.app/Contents/MacOS/lua/meta/art
lua debug: Trying Lua scripts in /Applications/VLC.app/Contents/MacOS/share/lua/meta/art
lua debug: Trying Lua playlist script /Applications/VLC.app/Contents/MacOS/share/lua/meta/art/00_musicbrainz.luac
lua debug: skipping script (unmatched scope) /Applications/VLC.app/Contents/MacOS/share/lua/meta/art/00_musicbrainz.luac
lua debug: Trying Lua playlist script /Applications/VLC.app/Contents/MacOS/share/lua/meta/art/01_googleimage.luac
lua debug: skipping script (unmatched scope) /Applications/VLC.app/Contents/MacOS/share/lua/meta/art/01_googleimage.luac
lua debug: Trying Lua playlist script /Applications/VLC.app/Contents/MacOS/share/lua/meta/art/02_frenchtv.luac
lua debug: skipping script (unmatched scope) /Applications/VLC.app/Contents/MacOS/share/lua/meta/art/02_frenchtv.luac
lua debug: Trying Lua playlist script /Applications/VLC.app/Contents/MacOS/share/lua/meta/art/03_lastfm.luac
lua debug: skipping script (unmatched scope) /Applications/VLC.app/Contents/MacOS/share/lua/meta/art/03_lastfm.luac
main debug: no art finder modules matched
main debug: looking for meta fetcher module matching "any": 1 candidates
lua debug: Trying Lua scripts in /Users/zander/Library/Application Support/org.videolan.vlc/lua/meta/fetcher
lua debug: Trying Lua scripts in /Applications/VLC.app/Contents/MacOS/lua/meta/fetcher
lua debug: Trying Lua scripts in /Applications/VLC.app/Contents/MacOS/share/lua/meta/fetcher
main debug: no meta fetcher modules matched
main debug: looking for art finder module matching "any": 2 candidates
lua debug: Trying Lua scripts in /Users/zander/Library/Application Support/org.videolan.vlc/lua/meta/art
lua debug: Trying Lua scripts in /Applications/VLC.app/Contents/MacOS/lua/meta/art
lua debug: Trying Lua scripts in /Applications/VLC.app/Contents/MacOS/share/lua/meta/art
lua debug: Trying Lua playlist script /Applications/VLC.app/Contents/MacOS/share/lua/meta/art/00_musicbrainz.luac
lua debug: Trying Lua playlist script /Applications/VLC.app/Contents/MacOS/share/lua/meta/art/01_googleimage.luac
lua debug: Trying Lua playlist script /Applications/VLC.app/Contents/MacOS/share/lua/meta/art/02_frenchtv.luac
lua debug: Trying Lua playlist script /Applications/VLC.app/Contents/MacOS/share/lua/meta/art/03_lastfm.luac
main debug: no art finder modules matched
live555 debug: connection error 404
live555 error: Failed to connect with rtsp://192.168.1.204:554/ch0_1.h264
main debug: no access_demux modules matched
main debug: creating access: rtsp://192.168.1.204/ch0_1.h264
main debug: looking for access module matching "rtsp": 23 candidates
satip debug: try to open 'rtsp://192.168.1.204/ch0_1.h264'
satip debug: connect to host '192.168.1.204'
main debug: net: connecting to 192.168.1.204 port 554
main debug: connection succeeded (socket = 16)
main debug: net: opening 0.0.0.0 datagram port 9502
main debug: net: opening 0.0.0.0 datagram port 9503
satip error: Failed to setup RTSP session
main debug: net: connecting to 192.168.1.204 port 554
main debug: connection succeeded (socket = 16)
access_realrtsp warning: Cseq mismatch, got 1, assumed 0
access_realrtsp debug: rtsp connected
access_realrtsp warning: only real/helix rtsp servers supported for now
main debug: no access modules matched
main debug: dead input
main debug: changing item without a request (current 13/14)
main debug: nothing to play
```

Does anybody have the same problem connectiong the low res stream?

Thanks, mzander

roleoroleo, 2021-01-05:
What model?

mzander, 2021-01-05:
Model is YYS.2016

mzander, 2021-01-05:
after settings a static ip address and several reboots it works now... thanks.

closed 2021-01-05 by mzander