roleoroleo/yi-hack-Allwinner-v2 #64: BFUS 9.0.19 snapshot no more working after changing default settings

rafal98, 2021-01-07:
I just receive 2 new Yi Home BFUS 9.0.19.

Operations done were:
1) configure Yi Home app to find the firmware version. Testing Video : ok
2) donwload firmware y21ga_0.1.3.tgz, and uncompress it on a 32GB fresh FAT32 SD
3) boot, flash, it's a success !! snapshot works
4) enable "disable cloud", enable "recording without cloud", disable all RTSP and all onvif features, disable FTP and telnet
5) saving (not rebooting)
6) disable "switch on/off camera" and save
7) configure FTP upload (server, path, user/pass) and save
8) rebooting using "maintenance".
9) snapshot return a 0 file size

using ssh:

```
root@yi-4a58:~# cd /tmp/sd/yi-hack/bin
root@yi-4a58:/tmp/sd/yi-hack/bin# ./imggrabber
Starting program
Could not open file /tmp/iframe.idx
root@yi-4a58:/tmp/sd/yi-hack/bin# ls -l /tmp
drwxr-xr-x    2 root     root            40 Jan  7 18:41 app
lrwxrwxrwx    1 root     root            23 Jan  7 18:41 audio -> /home/app/audio_file/us
-rw-r--r--    1 root     root             2 Jan  7 18:43 cloud_init_finish
-rw-r--r--    1 root     root            21 Jan  7 18:41 dns1
-rw-r--r--    1 root     root             2 Jan  7 18:41 done
drwxr-xr-x    2 root     root            40 Jan  7 18:41 face_detect
-rw-r--r--    1 root     root          2886 Jan  7 19:07 ftppush.log
-rw-r--r--    1 root     root            12 Jan  7 18:41 gw1
-rw-r--r--    1 root     root             0 Jan  7 18:41 httpd.conf
drwxr-xr-x    2 root     root            40 Jan  7 18:41 ko
drwxr-xr-x    2 root     root            40 Jan  7 18:41 log
-rw-r--r--    1 root     root            14 Jan  7 18:41 mask
-rwxr-xr-x    1 root     root          2644 Jan  7 18:41 mmap.info
-rw-r--r--    1 root     root           116 Jan  7 18:41 resolv.conf
drwxr-xr-x    2 root     root            40 Jan  7 18:41 run
drwxr-xr-x    8 root     root         32768 Jan  7 18:43 sd
drwxr-xr-x    2 root     root            40 Jan  7 18:41 tools
drwxr-xr-x    3 root     root            60 Jan  7 18:41 var
-rwxrwxrwx    1 root     root           191 Jan  7 18:41 wpa_supplicant.conf
```
I flash the 2nd camera, and copy the /tmp/iframe.idx to the first but it fail with another error:

```
root@yi-4a58:/tmp/sd/yi-hack/bin# ./imggrabber
Starting program
Mapping file /dev/shm/fshare_frame_buf, size 1786156, to b6cce000
Encoding h264 frame
Starting decode
Decode frame
[h264 @ 0x13c8b50] No start code is found.
[h264 @ 0x13c8b50] Error splitting the input into NAL units.
Error decoding frame
Error encoding h264 frame
```

Then I reset to the default config, reboot, but no change.
I tune the second camera, and after the reboot both camera can't see snaphot (so it's not an issue with an SD card for example)

I reformat SD card from first camera, putting the same firmware, rebooting and SNAPSHOT is OK !!

So I guess there is a hidden dependency in the configuration.

I can help you to do more test if you want

Regards

roleoroleo, 2021-01-08:
Why do you switch off the cam?
I mean "switch on/off camera" checkbox.

rafal98, 2021-01-08:
Hi, In fact I just want to do some "explicit" snapshot from domoticz (just check that light is off for example). I don't need to detect motion or record anything. I enable automatic detection only when I'm on holiday.

From what I wrote yesterday, I'm not so sure because today I format both SD, reapply the same firmware, but snapshot continue to fail on both camera (snapshot, and RTSP video are dead, but camera config is very responsive with a goog wifi quality)

rafal98, 2021-01-10:
About this error who generate or copy this missing file ?

root@(none):/tmp/sd/yi-hack/bin# ./imggrabber
Starting program
Could not open file /tmp/iframe.idx

rafal98, 2021-01-15:
After reinstalling and connecting using Yi app, camera appears off. So after put it online, I can again so some snapshot.

It's probably obvious to not be able to take snapshot if camera is off, but it's not the behaviour of the y25: camera off means no surveillance , but snapshot is possible. With the y21ga, if camera is off, surveillance and snapshot are disabled.

So I close the ticket

closed 2021-01-15 by rafal98

roleoroleo, 2021-01-16:
Now I understand.
In the y25 cam, the stream is picked up directly from the kernel driver.
If you switch off the cam, the stream continues to run.
In the allwineer based cams, the stream is read from the fshare_frame_buffer file (filled by rmm process).
If you switch off the cam, probably the rmm process doesn't fill the file.