roleoroleo/yi-hack-Allwinner-v2 #66: very poor quality

chrisdaloa, 2021-01-13:
Hi, I hacked my YI Home 1080P BFUSY441HW, firmware 9.0.19.02_202007141512 but the RTSP quality is very poor also in HIGH resolution mode. I see very large pixels, Is it normal?

domoson, 2021-01-13:
@chrisdaloa what viewer are you using? VLC?

chrisdaloa, 2021-01-14:
> 
> 
> @chrisdaloa what viewer are you using? VLC?

yes, vlc.

chrisdaloa, 2021-01-15:
@domoson any idea, pls?

domoson, 2021-01-15:
Can you confirm that quality without RTSP is better?

You can try to record a video, download the recorded video from cam (e.g. via included FTP server) and then compare the video to the live-quality, both in VLC.

Asking because these cameras are certainly not high-end, so maybe the quality you see is what is expected? Can you also share an example image or video showing the bad quality you mentioned?

roleoroleo, 2021-01-16:
The hack doesn't change the stream quality of the cam.
The RTSP server is fed with the same h264 stream that the cam sends to the app.
You could lose frames or have more delay but the quality should be the same.

closed 2022-06-11 by roleoroleo