roleoroleo/yi-hack-Allwinner-v2 #68: PTZ controls reversed

Zephael86, 2021-01-16:
On Yi 1080p Dome (r30gb) fw. version: 1.0.3 I see that the PTZ movement controls on the webpage are reversed.

When I rotate the image for ceiling mount, the controls move the camera as they should in floor mounting mode. And vica versa.

I see the same behavior when using Onvifer application.

roleoroleo, 2021-01-16:
Is this the only case?
Do the other users of the r30gb have the same problem?

fabiodirauso, 2021-01-24:
i have this problem, on the ptz command as invertited

roleoroleo, 2021-01-25:
Ok, I will change left with right.

Zephael86, 2021-01-25:
Up-Down is affected as well.
Without knowing the code I am just guessing. Can it be that it is invers stored that the cam is on its "foot" or ceil mounted?

roleoroleo, 2021-01-25:
I agree with you.

roleoroleo, 2021-01-25:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/0d0b7a50e674e095c1570153661fc5e577352189

closed 2022-06-11 by roleoroleo